andyboeh/esphome-elero #1: Example config.yaml

appi1, 2024-02-02:
hey
very interessting project. i use at the moment eleropy on micropython, but it is not realy easy to maintain, so the same on esphome plattform will be great.
do you have a complete config.yaml with the componet include line.
thanks

andyboeh, 2024-02-02:
As you might have noticed, this is incomplete and does not work (yet). Accordingly, there is no complete config. Stay tuned.

andyboeh, 2024-02-03:
@appi1 I was able to control my first blind tonight, so finally transmission works. However, the protocol is slightly different to the MicroPython version as I'm simulating my TempoTel2 - YMMV.

appi1, 2024-02-04:
@andyboeh  it works also for me, great job. the only point i had to change are the hex commands for my elero receiver ( step mode). i have to press the down button for 2 sec.  for a down command. only a short press  will move down, move up or til for only 2 cm. With the following code changes your code is working als for the step mode.
`static const uint8_t ELERO_COMMAND_COVER_CONTROL = 0x6a;
static const uint8_t ELERO_COMMAND_COVER_CHECK = 0x00;
static const uint8_t ELERO_COMMAND_COVER_STOP = 0x10;
static const uint8_t ELERO_COMMAND_COVER_UP = 0x21;  // changed from 0x20
static const uint8_t ELERO_COMMAND_COVER_TILT = 0x25;  //// changed from 0x24
static const uint8_t ELERO_COMMAND_COVER_DOWN = 0x41;  // changed from 0x40
static const uint8_t ELERO_COMMAND_COVER_INT = 0x45;   //// changed from 0x44
`

I tested also with mqtt and it works great

andyboeh, 2024-02-04:
Great, I'll make that (and a bunch of other things!) configurable.

appi1, 2024-02-06:
@andyboeh 
it works great, thank you. i tested with 10 blinds and mqtt, perfekt.
Can you please also implement the tilt an int(Zwischenposition) as configurable in the config.yaml
static const uint8_t ELERO_COMMAND_COVER_TILT = 0x25; //// changed from 0x24
static const uint8_t ELERO_COMMAND_COVER_INT = 0x45; //// changed from 0x44

andyboeh, 2024-02-06:
Yes and now: As of now, tilt isn't implemented as I do not have any covers that would support it. Implementing it seems doable but isn't trivial. There is a bunch of other stuff that I would like to fix/implement first (like position control), then I can have a go at this.

However, "Tilt" and "Zwischenposition" are two different things! When I implement position control, it should be possible to stop the cover in any desired position, not only in the pre-programmed intermediate position.

andyboeh, 2024-02-18:
From the perspective of the Cover component, there is only a target position and that is a value between 0.0 and 1.0, so there never was such a thing as "Up" or "Down". I have no experience with MQTT on ESPHome, could you post your configuration YAML?

appi1, 2024-02-18:
my config.yaml:

`#https://github.com/andyboeh/esphome-elero/tree/main

esphome:
  name: elero-esphome-2
  friendly_name: Elero_Esphome_2

esp32:
  board: lolin32_lite
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "5SA4KkkAYtHuyj8NgqRXvDNqgAI+2UTLgz4ZHmeliSE="

ota:
  password: "967417252603d37832366de08fe782e1"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Elero-Esphome-2 Fallback Hotspot"
    password: "Aa5GV21wgfXM"

mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password

  
# Example configuration entry
web_server:
  port: 80

captive_portal:

external_components:    # funktioniert mit lokaler Kopie  19.11.2023
  - source:
      type: local
      path: my_components/elero
    components: [elero]

#external_components:
#  - source: github://andyboeh/esphome-elero

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO19
  

elero:
  cs_pin: GPIO5
  gdo0_pin: GPIO17
  freq0: 0xc0
  freq1: 0x71
  freq2: 0x21

cover:
  - platform: elero
    blind_address: 0x9e881b
[... 132 lines cut ...]
[18:55:31][C][mqtt.cover:055]:   State Topic: 'elero-esphome-2/cover/kueche_west/state'
[18:55:31][C][mqtt.cover:055]:   Command Topic: 'elero-esphome-2/cover/kueche_west/command'
[18:55:31][C][mqtt.cover:057]:   Position State Topic: 'elero-esphome-2/cover/kueche_west/position/state'
[18:55:31][C][mqtt.cover:058]:   Position Command Topic: 'elero-esphome-2/cover/kueche_west/position/command'
[18:55:31][C][mqtt.cover:052]: MQTT cover 'Essen':
[18:55:31][C][mqtt.cover:055]:   State Topic: 'elero-esphome-2/cover/essen/state'
[18:55:31][C][mqtt.cover:055]:   Command Topic: 'elero-esphome-2/cover/essen/command'
[18:55:31][C][mqtt.cover:057]:   Position State Topic: 'elero-esphome-2/cover/essen/position/state'
[18:55:31][C][mqtt.cover:058]:   Position Command Topic: 'elero-esphome-2/cover/essen/position/command'
[18:55:31][C][mqtt.cover:052]: MQTT cover 'Stube_west':
[18:55:31][C][mqtt.cover:055]:   State Topic: 'elero-esphome-2/cover/stube_west/state'
[18:55:31][C][mqtt.cover:055]:   Command Topic: 'elero-esphome-2/cover/stube_west/command'
[18:55:31][C][mqtt.cover:057]:   Position State Topic: 'elero-esphome-2/cover/stube_west/position/state'
[18:55:31][C][mqtt.cover:058]:   Position Command Topic: 'elero-esphome-2/cover/stube_west/position/command'
[18:55:31][C][mqtt.cover:052]: MQTT cover 'Stube_ost':
[18:55:31][C][mqtt.cover:055]:   State Topic: 'elero-esphome-2/cover/stube_ost/state'
[18:55:31][C][mqtt.cover:055]:   Command Topic: 'elero-esphome-2/cover/stube_ost/command'
[18:55:31][C][mqtt.cover:057]:   Position State Topic: 'elero-esphome-2/cover/stube_ost/position/state'
[18:55:31][C][mqtt.cover:058]:   Position Command Topic: 'elero-esphome-2/cover/stube_ost/position/command'
[18:55:31][C][mqtt.cover:052]: MQTT cover 'Kueche_ost':
[18:55:31][C][mqtt.cover:055]:   State Topic: 'elero-esphome-2/cover/kueche_ost/state'
[18:55:31][C][mqtt.cover:055]:   Command Topic: 'elero-esphome-2/cover/kueche_ost/command'
[18:55:31][C][mqtt.cover:057]:   Position State Topic: 'elero-esphome-2/cover/kueche_ost/position/state'
[18:55:31][C][mqtt.cover:058]:   Position Command Topic: 'elero-esphome-2/cover/kueche_ost/position/command'
[18:55:31][C][mqtt.cover:052]: MQTT cover 'OG_Beschattung_Veranda_Kueche':

andyboeh, 2024-02-18:
I just checked the ESPHome code: If you send "OPEN" or "CLOSE" to the "command" topic, i.e. to 'elero-esphome-2/cover/kueche_west/command', it maps that to the internal positions 1.0 and 0.0. However, you can send positions as values to the topic 'elero-esphome-2/cover/kueche_west/position/command' (note the additional folder path). If you send 1.0 or 0.0 it should have the same effect as sending OPEN or CLOSE to the first "command" topic. I don't know how FHEM handles that, though.
If you are familiar with C++, this is defined at https://github.com/esphome/esphome/blob/ec3162282cd5a2c3455dd7a8669b4f762c481a20/esphome/components/mqtt/mqtt_cover.cpp#L18

andyboeh, 2024-02-21:
OK - can you give me a few details on how this is supposed to work with the original remote: 

1. Can you adjust it to any given til angle or just "full tilt" or so?
2. Do you need to press tilt repeatedly to get a desired angle?
3. Do you need to press tilt and stop it at the desired angle?
4. Can you tilt at any time or is it only supported in "full-close"?

Number 1 would be the easiest to implement. 2. would be a difficult, 3. feasible with time-based tilt similar to time-based position control.

appi1, 2024-02-21:
EXCELLENT,
There are two different receivers:

1: Combio III-868 JA /-915 for blind drives (corresponds to the roller shutter command)
command_up: 0x20
command_down: 0x40
and for tilt:
command_tilt: 0x24
and for Int:
command_int: 0x44
2: Combio III-868 JA **Pulse** /-915 for blind drives
I have this variant, which expects the following hex values:
command_up: 0x21
command_down: 0x41
and for tilt:
command_tilt: 0x25
and for Int:
command_int: 0x45
With the Combio III-868 JA Pulse, the turning path can be programmed in the receiver (via hand transmitter).
Actually I would need a command like up/down/stop for turning
So your suggested variant 1 would be correct. A command 0x41

Can you adjust it to any given til angle or just "full tilt" or something like that?
- the tilt angle can be programmed in the receiver, the transmitter is not interested
Do you need to press tilt repeatedly to get a desired angle?
- no
Do you need to press tilt and stop it at the desired angle?
- no
Can you tilt at any time or is it only supported in "full-close"?
- at every position where the blind is, except at the top

I hope it is understandable

andyboeh, 2024-02-21:
The intermediate position is a pre-programmed stop somewhere in the middle, right? I think that my motors support that too (RolTop-J 868), but we haven't programmed the intermediate position.

appi1, 2024-02-21:
yeh, right. it's easy to program the int postion, but as you wrote above it is no longer imortant with your positioning function.  But tilt is great in summer time
to ha no direct sun in the room and no darkness....

andyboeh, 2024-02-21:
Yeah, perfectly understandable (just not possible with my hard covers). I'll represent tilt with 0% and 100% then. Note that "tilt" could be the reason why you get wrong position information from the blinds: The tilt state isn't currently parsed and just ignored. The way I would like to implement it is: 
If you need tilt, you have to add the config option "tilt = True" and you will be able to specify the command to use. The cover then supports tilt and any target >50% will trigger the tilt command.

One more question: How do you "exit" tilt? I suppose by pressing "Up" or "Down", does it then really move up or down or just back to its non-tilted state?

appi1, 2024-02-21:
you don't have to implement a automatism for the tilt function. on the remote it is also a short button up press.
copy from user's manual with the pulse receiver:  sorry only german
`Betrieb des Senders mit Combio JA Pulse
Für Jalousieantriebe kann ein Combio-868/-915 JA Pulse zur
genauen Einstellung der Lamellen verwendet werden.
Bei einem Druck auf die AUF- oder AB-Taste wird die
voreingestellte Impulszeit vom Combio Pulse gefahren.
Die Impulszeit kann vom Anwender geändert werden. Dazu
an einem eingelernten Sender die STOPP- und AUF-Taste
6 Sekunden gedrückt halten. Der Antrieb beginnt in kleinen
Impulsen zu fahren. Sobald der Behang den gewünschten Fahrweg
zurückgelegt hat, die AUF-Taste loslassen, dann die STOPPTaste loslassen. Die neue Impulszeit ist gespeichert. Die neue
Impulszeit entspricht der Summe aller Impulszeiten während des
Programmierens der Impulszeit. Nach 30 Impulsen beendet der
Combio JA Pulse das Programmieren der Impulszeit.`

I think the short press of the Up Button will send the Hex 0c25, it is not a mode it's only a commant possible at every position of the blind. If a Tilt position is programmed the blind go up to the position and tilt but i think its the same caommand 0x25

appi1, 2024-02-21:
..... so we don't need the tilt enable, the tilt command 0x25 cames from the remote Variotel or from the esp32 if you want tilt at the curret position or from bottom up on the programmed position.

andyboeh, 2024-02-21:
OK, but we are still limited by what ESPhome / Home Assistant provides for covers and we need to somehow map that. Tilt position is, similar to the cover position, represented by a value between 0.0 and 1.0 so we need to use that interface. Hence, I can send out 0x25 if tilt >0.5 is requested and nothing (?) if tilt < 0.5 is requested (or the other way round). And enable is required, otherwise all Elero covers have the tilt feature added to the Home Assistant UI which is not what I want.

andyboeh, 2024-02-21:
Experimental support was added in 4108f0c. I still do not know how to reset tilt back to normal, so this is currently not implemented. Any value >0 for tilt should send out the tilt command.
You will need these two options:
```
supports_tilt: True
tilt_command: 0x25
```

appi1, 2024-02-22:
hey, it works absolutely perfectly. You dont have to reset tilt to normal, because Tilt is only a small movement "Up" of the blind maybe for for 3 cm ( as programmed in the Receiver ). So position of the blind neary the same and the blind waits fo an new command. 

ps: in your last comment at the end tilt_command: 0x24 is correct

andyboeh, 2024-02-22:
Well, I'm referring to the representation in HA here: If tilt is not reset, it's always shown as 100% tilt even if the blind is moving or in top position. For the time being, the internal state is reset whenever a new up/down command is sent (which makes sense). Basically, you can enter tilt by setting a position > 0, but you can't leave it without up/down - setting tilt to 0 has no effect.

Great that it's working :)

closed 2024-12-19 by andyboeh