andyboeh/esphome-elero #2: Can't get blinds to move

breinsp, 2024-03-30:
Hi, 

first of all: really cool project!

I have sucessfully received packets and they look like this in my home (up command):

`len=27, cnt=186, typ=0x44, typ2=0x00, hop=00, syst=01, chl=03, src=0x86ab10, bwd=0x86ab10, fwd=0x86ab10, #dst=01, dst=000003, rssi=-51.0, lqi=45, crc= 1, payload=[0x00 0x04 0x00 0x00 0x20 0x00 0x00 0x00 0x00 0x40]`

However, I can't get the blinds to move. These are the messages I am sending:

`len=29, cnt=01, typ=0x44, typ2=0x00, hop=00, syst=01, chl=03, src=0x86ab10, bwd=0x86ab10, fwd=0x86ab10, #dst=01, dst=000003, payload=[0x00 0x04 0x8f 0x71 0x20 0x00 0x00 0x00 0x00 0x00]`

Everything seems to match, except payload's 3rd, 4th and 10th byte - any idea what could be the issue?

Also the readme seems to be incorrect regarding the index of the command byte:
`The very last byte of the payload doesn't matter, the second-last byte is the command to send. `
should be the 5th byte.

This is my config:

```
external_components:
  - source: github://breinsp/esphome-elero
    refresh: 0s

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO19

elero:
  cs_pin: GPIO5
  gdo0_pin: GPIO17
  freq0: 0x7a
  freq1: 0x71
  freq2: 0x21

cover:
  - platform: elero
    blind_address: 0x000003
    channel: 3
    remote_address: 0x86ab10
    name: Bedroom
    payload_1: 0x00
    payload_2: 0x04
    pck_inf1: 0x44
    pck_inf2: 0x00
    hop: 0x00
    ```

Thanks!

andyboeh, 2024-03-31:
I think that the packet you decoded is not a command, it's too short. I see such packets too, although with packet type 0x45. I do not know what they are, the source address doesn't match any of my blinds.

Try again to capture an actual command - maybe you need to adjust the frequency of your CC1101 module?

> Also the readme seems to be incorrect regarding the index of the command byte:

You are right. Something's wrong with my example, thanks.

breinsp, 2024-03-31:
You are right, I was looking at the wrong packets.

I fixed it by focussing on the ones with len=29 and got the correct config - now they're moving 🥳 

Thank you very much, appreciate the help!

closed 2024-03-31 by breinsp