andyboeh/esphome-elero #3: Unable to send command despite clean reading from remote

RadekJurzysta, 2024-04-22:
Hello,
I have two elero blinds that I'd like to control using your solution. The blinds are currently controlled via VarioCom Slide remote, with the blinds assigned to bank 1 & 2. For the purpose of the testing I'm sticking to #1
I'm using a module that looks like this one:
[image]

Command reading seems to be working fine, but I'm unable to make the blinds move by sending out commands.
Also I noticed a strange behaviour when I send command. The below lines are looping, until I press STOP:
`[12:26:54][W][component:237]: Component elero.cover took a long time for an operation (55 ms).
[12:26:54][W][component:238]: Components should block for at most 30 ms.
[12:26:55][D][cover:170]: 'Salon' - Publishing:
[12:26:55][D][cover:173]:   Position: 100%
[12:26:55][D][cover:184]:   Tilt: 0%
[12:26:55][D][cover:186]:   Current Operation: OPENING`

UP command reading log result:
`[D][elero:422]: len=27, cnt=222, typ=0x44, typ2=0x10, hop=00, syst=01, chl=01, src=0x1c3e55, bwd=0x1c3e55, fwd=0x1c3e55, #dst=01, dst=000001, rssi=-20.0, lqi=127, crc= 1, payload=[0x00 0x03 0x00 0x00 0x20 0x00 0x00 0x00 0x00 0x40]`

My full ESPHome config:
`substitutions:
  name: esphome-web-67337c
  friendly_name: Zaluzje

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

external_components:
  - source: github://andyboeh/esphome-elero

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO19

elero:
  cs_pin: GPIO17
  gdo0_pin: GPIO26
  freq0: 0xc0
  freq1: 0x71
  freq2: 0x21

cover:
  - platform: elero
    blind_address: 0x000001
    channel: 1
    remote_address: 0x1c3e55
    name: Salon
    open_duration: 45s
    close_duration: 46s
    poll_interval: 1min
    supports_tilt: true
    payload_1: 0x00
    payload_2: 0x03
    pck_inf1: 0x44
    pck_inf2: 0x10
    hop: 0x00
    command_check: 0x00
    command_stop: 0x10
    command_up: 0x20
    command_down: 0x40
    command_tilt: 0x24`

RadekJurzysta, 2024-04-22:
As a quick follow up: is there a way to check whether all of my wires are correctly connected?

andyboeh, 2024-04-22:
This is not the UP command you are seeing, it's too short (len=27 should be len=29). Check again.

andyboeh, 2024-04-22:
I'll make that clearer in the README :)