andyboeh/esphome-elero #5: Blinds sometimes not moving

peterjuras, 2024-07-07:
Hi!

First of all thank you for creating this great component, I really appreciate it! 🙏

I'm using your component with three elero blinds. I have automations that simultaneously close & open three blinds when the sun goes down & up, but I noticed that sometimes one of the blinds will not move.

It also doesn't move when I manually & repeatedly send the command again a few times after a couple of seconds. If I wait a couple of minutes it typically works again.

Do you know if there is an issue with simultaneous control of blinds? Should I wait a certain time between sending commands or do you have other recommendations? I don't think it is a range issue, since it works most of the time and it is the "middle" blind and even closest to my ESP32.

Thanks,
Peter

andyboeh, 2024-07-07:
Actually, I'm seeing the same issue on the single blind that I control automatically. I thought it was due to the distance, I cannot get a direct connection to the blind. 
Do you get any errors in the log?

peterjuras, 2024-07-07:
I haven't noticed any errors in my logs last time I checked, but I'll try to monitor for logs again if it happens again.

Do you also control multiple blinds at the same time?

andyboeh, 2024-07-07:
No, not at this time, it's likely that there are some problems here.

peterjuras, 2024-07-08:
I'll check today to see if there are any logs during a simultaneous send of closing covers 👍

peterjuras, 2024-07-08:
I checked for logs today. All three blinds did move successfully, however there were still a couple of error messages in the logs:

```
INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/living-room-elero-transmitter.yaml...
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 192.168.68.72 using esphome API
INFO Successfully connected to living-room-elero-transmitter @ 192.168.68.72 in 0.023s
INFO Successful handshake with living-room-elero-transmitter @ 192.168.68.72 in 0.088s
[21:44:50][I][app:100]: ESPHome version 2024.6.6 compiled on Jul  7 2024, 03:10:59
[21:44:50][C][wifi:599]: WiFi:
[21:44:50][C][wifi:427]:   Local MAC: D4:D4:DA:E3:C4:E0
[21:44:50][C][wifi:432]:   SSID: 'Rutzheim'[redacted]
[21:44:50][C][wifi:435]:   IP Address: 192.168.68.72
[21:44:50][C][wifi:439]:   BSSID: 96:53:22:82:E3:10[redacted]
[21:44:50][C][wifi:440]:   Hostname: 'living-room-elero-transmitter'
[21:44:50][C][wifi:442]:   Signal strength: -81 dB ▂▄▆█
[21:44:50][C][wifi:446]:   Channel: 3
[21:44:50][C][wifi:447]:   Subnet: 255.255.252.0
[21:44:50][C][wifi:448]:   Gateway: 192.168.68.1
[21:44:50][C][wifi:449]:   DNS1: 1.1.1.1
[21:44:50][C][wifi:450]:   DNS2: 8.8.8.8
[21:44:50][C][logger:185]: Logger:
[21:44:50][C][logger:186]:   Level: DEBUG
[21:44:50][C][logger:188]:   Log Baud Rate: 115200
[21:44:50][C][logger:189]:   Hardware UART: UART0
[21:44:50][C][spi:068]: SPI bus:
[21:44:50][C][spi:069]:   CLK Pin: GPIO18
[21:44:50][C][spi:070]:   SDI Pin: GPIO19
[21:44:50][C][spi:071]:   SDO Pin: GPIO23
[21:44:50][C][spi:076]:   Using HW SPI: SPI2_HOST
[21:44:50][C][elero:046]: Elero Config: 
[21:44:50][C][elero.cover:012]: Elero Cover 'Living room balcony door'
[21:44:50][C][elero.cover:012]:   Assumed State: YES
[21:44:50][C][elero.cover:012]: Elero Cover 'Living room window'
[21:44:50][C][elero.cover:012]:   Assumed State: YES
[21:44:50][C][elero.cover:012]: Elero Cover 'Dining room'
[21:44:50][C][elero.cover:012]:   Assumed State: YES
[21:44:50][C][mdns:115]: mDNS:
[21:44:50][C][mdns:116]:   Hostname: living-room-elero-transmitter
[21:44:50][C][esphome.ota:073]: Over-The-Air updates:
[21:44:50][C][esphome.ota:074]:   Address: living-room-elero-transmitter.local:3232
[21:44:50][C][esphome.ota:075]:   Version: 2
[21:44:50][C][esphome.ota:078]:   Password configured
[21:44:50][C][safe_mode:018]: Safe Mode:
[21:44:50][C][safe_mode:020]:   Boot considered successful after 60 seconds
[21:44:50][C][safe_mode:021]:   Invoke after 10 boot attempts
[21:44:50][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[21:44:50][C][api:139]: API Server:
[21:44:50][C][api:140]:   Address: living-room-elero-transmitter.local:6053
[21:44:50][C][api:142]:   Using noise encryption: YES
[21:45:02][D][elero:451]: rcv'd: len=29, cnt=119, typ=0xca, typ2=0x10, hop=00, syst=01, chl=153, src=0xac601d, bwd=0xac601d, fwd=0x4daf0c, #dst=01, dst=4daf0c, rssi=-83.0, lqi=46, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x10]
[21:45:03][D][elero:451]: rcv'd: len=29, cnt=54, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=146, src=0xa8ab21, bwd=0xa8ab21, fwd=0x4daf0c, #dst=01, dst=4daf0c, rssi=-82.0, lqi=47, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x90]
[21:46:20][D][elero:451]: rcv'd: len=29, cnt=58, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=09, src=0x99601d, bwd=0x99601d, fwd=0x4daf0c, #dst=01, dst=4daf0c, rssi=-88.0, lqi=45, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x90]
[21:46:20][D][elero:451]: rcv'd: len=29, cnt=59, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=09, src=0x99601d, bwd=0x99601d, fwd=0x4daf0c, #dst=01, dst=4daf0c, rssi=-88.0, lqi=51, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x90]
[21:50:02][D][elero:451]: rcv'd: len=29, cnt=120, typ=0xca, typ2=0x10, hop=00, syst=01, chl=154, src=0xac601d, bwd=0xac601d, fwd=0x4daf0c, #dst=01, dst=4daf0c, rssi=-83.0, lqi=48, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x10]
[21:50:03][D][elero:451]: rcv'd: len=29, cnt=56, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=147, src=0xa8ab21, bwd=0xa8ab21, fwd=0x4daf0c, #dst=01, dst=4daf0c, rssi=-83.0, lqi=46, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x90]
[21:50:03][D][elero:451]: rcv'd: len=29, cnt=57, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=147, src=0xa8ab21, bwd=0xa8ab21, fwd=0x4daf0c, #dst=01, dst=4daf0c, rssi=-83.0, lqi=45, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x10]
[21:51:20][D][elero:451]: rcv'd: len=29, cnt=61, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=10, src=0x99601d, bwd=0x99601d, fwd=0x4daf0c, #dst=01, dst=4daf0c, rssi=-90.0, lqi=47, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x10]
[21:55:02][D][elero:451]: rcv'd: len=29, cnt=123, typ=0xca, typ2=0x10, hop=00, syst=01, chl=155, src=0xac601d, bwd=0xac601d, fwd=0x4daf0c, #dst=01, dst=4daf0c, rssi=-83.0, lqi=45, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x90]
[21:55:03][D][elero:451]: rcv'd: len=29, cnt=59, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=148, src=0xa8ab21, bwd=0xa8ab21, fwd=0x4daf0c, #dst=01, dst=4daf0c, rssi=-82.0, lqi=47, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x90]
[... cut ...]
[22:18:29][D][cover:173]:   Position: 15%
[22:18:29][D][cover:186]:   Current Operation: CLOSING
[22:18:29][D][elero:451]: rcv'd: len=29, cnt=156, typ=0xca, typ2=0x10, hop=00, syst=01, chl=174, src=0xac601d, bwd=0xac601d, fwd=0x4daf0c, #dst=01, dst=4daf0c, rssi=-82.0, lqi=47, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x0d 0x90]
[22:18:58][D][esp32.preferences:114]: Saving 3 preferences to flash...
[22:18:58][D][esp32.preferences:143]: Saving 3 preferences to flash: 1 cached, 2 written, 0 failed

```

andyboeh, 2024-07-09:
I've never experienced those "Error transferring, 28 bytes left in buffer" message before. I can try to increase the timeout a bit, I'm not sure how long it *really* takes to transfer everything. 

What I have seen is the "Timed out waiting for TX" message, although I'm not sure for which reason.

I'll have a look, but it's going to take a couple of days.

peterjuras, 2024-07-10:
Thanks, a lot for looking into it! It works correctly most of the times, therefore it might also be unrelated to this issue.

peterjuras, 2024-07-20:
Hi @andyboeh ,

I tried to play around with the number of timeouts, packages to send & retries but sadly it still appears to error out (`Error transferring, 28 bytes left in buffer`) sometimes and some commands are either not sent or make it to the blinds. Do you have any additional hints on what could be wrong?

andyboeh, 2024-07-29:
atm, every request is sent twice within a very short time span. Have you tried setting `ELERO_SEND_PACKETS` to 1 or have you tried increasing the delay between the packets, i.e. `ELERO_DELAY_SEND_PACKETS`?
And I can see a design problem with how commands are handled: There is currently one loop per cover that is responsible for sending commands and keeping track of the timeouts. When you have several blinds, it might happen that commands from different blinds are sent very, very frequently.

I **think** that there needs to be one loop responsible for sending commands for the whole component, not per cover. And then it would be necessary to block polling covers when a cover is moving so that we don't block the radio.

peterjuras, 2024-08-10:
Hi @andyboeh ,
yes I tried playing around with `ELERO_SEND_PACKETS` & `ELERO_DELAY_SEND_PACKETS` - both decreasing and increasing the values (independently). It did not really improve the situation, but sometimes made it worse.

dcferreira, 2026-10-12:
I had this problem for ages and recently gave it another go at trying to solve it and managed for my particular case!

Same general symptoms: blinds randomly fail to move. There was no strong correlation between distance to the radio and how often a blind failed, so I was actually convinced that the signal strength was not the cause. There was one exception though: one blind the farthest away from the radio rarely moved.

The biggest reason why I thought it couldn't be the signal strength was because I'd read in the manual that the blinds themselves repeat the signal, so it didn't make sense for one to fail while the other was working. It turns out while this capability exists it wasn't turned on in my case!

So 2 fixes that fixed my issues:
1. `pck_inf1: 0x6a` (the default in this repo) disables having the blinds relaying the signal. Changing that value to `0x69` in my case turned it on.
2. Still with that turned on not all packets get relayed immediately, so I updated the code to send each command twice and re-send if no status reply arrives within 2s

Right now the blinds have been 100% reliable for almost a week. There is a small downside that sometimes they take a while to respond (often like 1-2s, in rare occasions like 10s). The farthest blind from the radio, which was rarely working before, now works reliably.

If it's helpful, [here](https://github.com/andyboeh/esphome-elero/compare/main...dcferreira:esphome-elero:dcferreira/debug/unreliable-commands) is my branch. Look for the latest commits with claude, that was my debugging session and where I reached the learnings above.

Hope that's helpful to someone!

andyboeh, 2026-10-12:
It's the `hop` parameter that controls relaying under normal circumstances. 0x00 disables relaying. It's unknown what `pck_inf1` does, so it might have the same effect.

There is already a send-retry (all packets are sent twice and three retries are made if no answer is received). This is explained two posts above yours.

dcferreira, 2026-10-13:
Yeah that's what I meant, I had `hop` at `0x0a` and the blinds still weren't relaying the signal until I changed `pck_inf1`. Not sure why my blinds are different to yours, but that's what I'm seeing 🤷 

Indeed your code already sends twice (which I'd disabled months ago in my fork while debugging, though I don't remember why), but as far as I can tell it doesn't re-send if no answer is received. It does retry a 3rd time if the radio itself reports the sending failed, but if the sending succeeds and nothing hears it, it doesn't retry as far as I can tell.

Thanks a lot for the repo @andyboeh and all the documentation in it btw! I realise you don't have the wireless blinds anymore so this discussion isn't very helpful to you, but if someone else has similar problems I'm happy to give more details