andyboeh/esphome-elero #6: MonoTel compatible

Nick75wien, 2024-08-16:
Hello!

I found your project and it might be a solution to control my Schlotterer window rollos. 
I have a MonoTel remote control by elero and so I wonder if your solution is compatible with that?

KR Niki

andyboeh, 2024-08-19:
Short answer: It depends. 
Long answer: There are two Elero protocols supported by your MonoTel. Only the bidirectional one works. Please check the manual: file:///home/andy/Downloads/MonoTel%202%20-%20Operating%20instruction%20-%20Radio%20hand-held%20transmitters.pdf

Nick75wien, 2024-08-19:
Thanks, but I can‘t see the pdf

andyboeh, 2024-08-19:
Sorry, I accidentally copied the wrong link - fixed.

Nick75wien, 2024-08-19:
Thanks! My remote looks really old in contrast to that:
[image]
[image]

andyboeh, 2024-08-19:
Uh... I don't know what protocol those speak, probably the old, unsupported one. If the blinds are older than ca. 10 years, it's the old protocol.

Nick75wien, 2024-08-31:
now that I have the hardware and some time to play arround I wonder if I need to install anything elese on Home Assistant or just flash the ESP32 with your code ?

andyboeh, 2024-08-31:
Just flash it with ESPHome and the custom component. Home Assistant will automatically pick up the covers once you add the ESPHome device.

closed 2024-12-19 by andyboeh