andyboeh/esphome-elero #7: Lost in the jungle. Need some help.

Hufenbacke, 2024-09-06:


andyboeh, 2024-09-07:
Please post the complete log. I updated your comment to include proper quoting, you have to use three accents, like so: "```"

Hufenbacke, 2024-09-07:
> Please post the complete log. I updated your comment to include proper quoting, you have to use three accents, like so: "```"

Thanks for you reply. Didn´t know it had to be "```".
Here is the full log:
```
INFO ESPHome 2024.8.3
INFO Reading configuration /config/esphome/elerosteuereinheit.yaml...
WARNING GPIO5 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from 10.20.20.80 using esphome API
INFO Successfully connected to elerosteuereinheit @ 10.20.20.80 in 0.071s
INFO Successful handshake with elerosteuereinheit @ 10.20.20.80 in 0.109s
[09:20:18][I][app:100]: ESPHome version 2024.8.3 compiled on Sep  7 2024, 09:11:39
[09:20:18][C][wifi:600]: WiFi:
[09:20:18][C][wifi:428]:   Local MAC: *********
[09:20:18][C][wifi:433]:   SSID: '*******'[redacted]
[09:20:18][C][wifi:436]:   IP Address: 10.20.20.80
[09:20:18][C][wifi:440]:   BSSID: ********[redacted]
[09:20:18][C][wifi:441]:   Hostname: 'elerosteuereinheit'
[09:20:18][C][wifi:443]:   Signal strength: -81 dB ▂▄▆█
[09:20:18][C][wifi:447]:   Channel: 7
[09:20:18][C][wifi:448]:   Subnet: 255.255.255.0
[09:20:18][C][wifi:449]:   Gateway: 10.20.20.1
[09:20:18][C][wifi:450]:   DNS1: 10.20.20.21
[09:20:18][C][wifi:451]:   DNS2: 10.20.20.22
[09:20:18][C][logger:185]: Logger:
[09:20:18][C][logger:186]:   Level: DEBUG
[09:20:18][C][logger:188]:   Log Baud Rate: 115200
[09:20:18][C][logger:189]:   Hardware UART: UART0
[09:20:18][C][spi:064]: SPI bus:
[09:20:18][C][spi:065]:   CLK Pin: GPIO18
[09:20:18][C][spi:066]:   SDI Pin: GPIO19
[09:20:18][C][spi:067]:   SDO Pin: GPIO23
[09:20:18][C][spi:072]:   Using HW SPI: SPI
[09:20:18][C][elero:046]: Elero Config: 
[09:20:18][C][elero.cover:012]: Elero Cover 'Schlafzimmer'
[09:20:18][C][elero.cover:012]:   Assumed State: YES
[09:20:18][C][captive_portal:088]: Captive Portal:
[09:20:18][C][mdns:116]: mDNS:
[09:20:18][C][mdns:117]:   Hostname: elerosteuereinheit
[09:20:18][C][esphome.ota:073]: Over-The-Air updates:
[09:20:18][C][esphome.ota:074]:   Address: 10.20.20.80:3232
[09:20:18][C][esphome.ota:075]:   Version: 2
[09:20:18][C][esphome.ota:078]:   Password configured
[09:20:18][C][safe_mode:018]: Safe Mode:
[09:20:18][C][safe_mode:020]:   Boot considered successful after 60 seconds
[09:20:18][C][safe_mode:021]:   Invoke after 10 boot attempts
[09:20:18][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[09:20:18][C][api:139]: API Server:
[09:20:18][C][api:140]:   Address: 10.20.20.80:6053
[09:20:18][C][api:142]:   Using noise encryption: YE
```

andyboeh, 2024-09-07:
OK, that looks fine - the wiring of your CC1101 module matches the configured pins?

Hufenbacke, 2024-09-07:
> OK, that looks fine - the wiring of your CC1101 module matches the configured pins?

I already checked it like 50times or so :)
I also took a multimeter to test the connection from all the the pins of the esp32 to the pins of the processor of the c1101. And all the connections work.

Btw: Was my guess right, that the communication log should appear in the esphomeweb console or do i have to look somewhere else?

andyboeh, 2024-09-07:
It should also appear in the web console. Try to send anything from the fake cover, i.e. try to open and close it. What is reported in the log?

Hufenbacke, 2024-09-07:
> It should also appear in the web console. Try to send anything from the fake cover, i.e. try to open and close it. What is reported in the log?

The log:
```
[10:44:52][C][api:139]: API Server:
[10:44:52][C][api:140]:   Address: 10.20.20.80:6053
[10:44:52][C][api:142]:   Using noise encryption: YES
[10:45:12][D][cover:076]: 'Schlafzimmer' - Setting
[10:45:12][D][cover:084]:   Position: 100%
[10:45:12][D][cover:170]: 'Schlafzimmer' - Publishing:
[10:45:12][D][cover:173]:   Position: 50%
[10:45:12][D][cover:186]:   Current Operation: OPENING
[10:45:12][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:12][D][elero.cover:086]: Retry #0 for blind 0xa831e5
[10:45:12][D][cover:170]: 'Schlafzimmer' - Publishing:
[10:45:12][D][cover:173]:   Position: 50%
[10:45:12][D][cover:186]:   Current Operation: OPENING
[10:45:12][W][component:237]: Component elero.cover took a long time for an operation (77 ms).
[10:45:12][W][component:238]: Components should block for at most 30 ms.
[10:45:12][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:12][D][elero.cover:086]: Retry #1 for blind 0xa831e5
[10:45:12][W][component:237]: Component elero.cover took a long time for an operation (57 ms).
[10:45:12][W][component:238]: Components should block for at most 30 ms.
[10:45:12][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:12][D][elero.cover:086]: Retry #2 for blind 0xa831e5
[10:45:12][W][component:237]: Component elero.cover took a long time for an operation (57 ms).
[10:45:12][W][component:238]: Components should block for at most 30 ms.
[10:45:12][E][elero:436]: Received invalid packet: dests_len too long (125)
[10:45:12][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:12][D][elero.cover:086]: Retry #3 for blind 0xa831e5
[10:45:12][E][elero.cover:089]: Hit maximum number of retries, giving up.
[10:45:12][W][component:237]: Component elero.cover took a long time for an operation (63 ms).
[10:45:12][W][component:238]: Components should block for at most 30 ms.
[10:45:12][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:12][D][elero.cover:086]: Retry #0 for blind 0xa831e5
[10:45:12][W][component:237]: Component elero.cover took a long time for an operation (57 ms).
[10:45:12][W][component:238]: Components should block for at most 30 ms.
[10:45:12][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:12][D][elero.cover:086]: Retry #1 for blind 0xa831e5
[10:45:12][W][component:237]: Component elero.cover took a long time for an operation (57 ms).
[10:45:12][W][component:238]: Components should block for at most 30 ms.
[10:45:12][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:12][D][elero.cover:086]: Retry #2 for blind 0xa831e5
[10:45:12][W][component:237]: Component elero.cover took a long time for an operation (56 ms).
[10:45:12][W][component:238]: Components should block for at most 30 ms.
[10:45:12][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:12][D][elero.cover:086]: Retry #3 for blind 0xa831e5
[10:45:12][E][elero.cover:089]: Hit maximum number of retries, giving up.
[10:45:12][W][component:237]: Component elero.cover took a long time for an operation (61 ms).
[10:45:12][W][component:238]: Components should block for at most 30 ms.
[10:45:13][D][cover:170]: 'Schlafzimmer' - Publishing:
[10:45:13][D][cover:173]:   Position: 54%
[10:45:13][D][cover:186]:   Current Operation: OPENING
[10:45:14][D][cover:076]: 'Schlafzimmer' - Setting
[10:45:14][D][cover:080]:   Command: STOP
[10:45:14][D][cover:170]: 'Schlafzimmer' - Publishing:
[10:45:14][D][cover:173]:   Position: 58%
[10:45:14][D][cover:186]:   Current Operation: IDLE
[10:45:14][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:14][D][elero.cover:086]: Retry #0 for blind 0xa831e5
[10:45:14][W][component:237]: Component elero.cover took a long time for an operation (58 ms).
[10:45:14][W][component:238]: Components should block for at most 30 ms.
[10:45:14][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:14][D][elero.cover:086]: Retry #1 for blind 0xa831e5
[10:45:14][W][component:237]: Component elero.cover took a long time for an operation (57 ms).
[10:45:14][W][component:238]: Components should block for at most 30 ms.
[10:45:14][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:14][D][elero.cover:086]: Retry #2 for blind 0xa831e5
[10:45:14][W][component:237]: Component elero.cover took a long time for an operation (58 ms).
[10:45:14][W][component:238]: Components should block for at most 30 ms.
[... 50 lines cut ...]
[10:45:16][D][cover:186]:   Current Operation: CLOSING
[10:45:16][D][cover:076]: 'Schlafzimmer' - Setting
[10:45:16][D][cover:080]:   Command: STOP
[10:45:16][D][cover:170]: 'Schlafzimmer' - Publishing:
[10:45:16][D][cover:173]:   Position: 51%
[10:45:16][D][cover:186]:   Current Operation: IDLE
[10:45:16][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:16][D][elero.cover:086]: Retry #0 for blind 0xa831e5
[10:45:16][W][component:237]: Component elero.cover took a long time for an operation (59 ms).
[10:45:16][W][component:238]: Components should block for at most 30 ms.
[10:45:16][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:16][D][elero.cover:086]: Retry #1 for blind 0xa831e5
[10:45:16][W][component:237]: Component elero.cover took a long time for an operation (57 ms).
[10:45:16][W][component:238]: Components should block for at most 30 ms.
[10:45:17][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:17][D][elero.cover:086]: Retry #2 for blind 0xa831e5
[... cut ...] or an operation (57 ms).
[10:45:17][W][component:238]: Components should block for at most 30 ms.
[10:45:17][E][elero:186]: Timed out waiting for TX: 0x0d
[10:45:17][D][elero.cover:086]: Retry #3 for blind 0xa831e5
[10:45:17][E][elero.cover:089]: Hit maximum number of retries, giving up.
[10:45:17][W][component:237]: Component elero.cover took a long time for an operation (60 ms).
[10:45:17][W][component:238]: Components should block for at most 30 ms.
```
It´s is still retrying as of writing :)

andyboeh, 2024-09-07:
Yes, you received something, but the data is complete nonsense. 182 destination blinds?
The message "Timed out waiting for TX" should not appear. It stays in state "0x0d" which is Rx, so your module never enters the transmit state. This can have two reasons:
1. The module does CCA first (Clear Channel Assessment). If CCA fails (i.e. the channel is in use), then the module statys in Rx.
2. Your module behaves differently than mine
3. The module is broken

Which CC1101 module are you using?

Hufenbacke, 2024-09-07:
> Yes, you received something, but the data is complete nonsense. 182 destination blinds? The message "Timed out waiting for TX" should not appear. It stays in state "0x0d" which is Rx, so your module [...]

I found my mistake. I didn´t know that you have to connect VCC and GND too. Very obvious know that I know it, but for a first timer. I just did what you wrote in the yaml. I can capture packets now. But because of the meshing, I don´t know the right adress and channel yet.

andyboeh, 2024-09-07:
That's a good one :) VCC and GND is so obvious that I did not explicitly state that.

Hufenbacke, 2024-09-08:
> That's a good one :) VCC and GND is so obvious that I did not explicitly state that.

Just to let you know. I got everything working. Maybe I will come back with a more specific question, if one arises at a later date, but for know, I am going to enjoy moving eleros up and down through HA. Thanks for your help!

closed 2024-09-08 by Hufenbacke