andyboeh/esphome-elero #9: Freq offset settings

ggggh, 2024-10-08:
I'm getting very very bad reception, as in the Tempotel2 literally needs to touch the antenna and even then only a few of the messages are received.    I assume I need to tune the freq offsets, but, how did you come up with your specific values and how would I check which are appropriate to me, other than trying out different combinations?

andyboeh, 2024-10-08:
The offsets are directly taken from the Tempo Tel 2, this is how the CC1101 in the TempoTel 2 is configured. My module works better with the following configuration:

```
  freq0: 0xc0
  freq1: 0x71
  freq2: 0x21
```

I usually use an RTL-SDR device to compare the frequency sent by the original remote to the frequency sent by my modules.

closed 2024-10-08 by andyboeh

andyboeh, 2024-10-08:
Ah, sorry, I misunderstood the question then: It's explained in the datasheet. I use an old AskSinPP Excel helper to calculate the values: https://github.com/TomMajor/SmartHome/tree/master/Info/CC1101_Frequenz