andyboeh/esphome-elero #10: Controlling channels with more than one blinds

hagsson, 2024-10-19:
Hi,
first of all a big thank you for this great project. I have set it all up and it's working great.
One issue I find myself with:
I have on my TempoTel2 two covers allocated to the same channel (in order to make it work with 10). When I control the channel via the code, it only operates one of the covers.
When I actuate the channel via tempotel, it works with both, eventhough it indicates a different len value when I read the communication: Len 32 instead of 29 for this. 

I tried following fix: I have buttons for the windows that also can operate the covers and lower them based on the measured light intensity. I tried to read the communication values in the code and then add it in my programmed remote. However, it also did have a different len value (27 instead of 29) and it did not work (no control possible).

I assume it's due to the different Len value that this does not work. Do you have any idea on how to fix this?

Thanks!

andyboeh, 2024-10-20:
Channel groups / multiple destinations are not supported. You need to define two individual blinds for it. I suppose that your problem is getting the address of the other blind: You need to change the function "interpret_msg" in elero.cpp to decode more than one destination as only the first destination is currently visible.

hagsson, 2024-10-28:
Tha ja a lot!

closed 2024-10-28 by hagsson