andyboeh/esphome-elero #11: Weird payload used in sent message

sidious38, 2024-10-26:
Thank you very much for implementing this project; I came across it after a long research regarding my Weinor blinds...
As described in the instructions, I obtained and soldered a C1101 radio module for 868MHz and can receive all commands. Unfortunately, sending doesn’t work as expected:

In the payload, I noticed that it does not match the received payload. Is this intentional? Shouldn’t the message simply be "copied," or did I misunderstand the concept?

```
[22:50:44][VV][elero:474]: send_command called
[22:50:44][V][elero:504]: send: len=29, cnt=03, typ=0x6a, typ2=0x10, hop=0a, syst=01, chl=17, src=0x82a501, bwd=0x82a501, fwd=0x82a501, #dst=01, dst=0xa46d20, payload=[0x00 0x03 0xc3 0xd6 0x2f 0x3a 0x49 0x85 0xc3 0x31]
[22:50:44][VV][elero:206]: transmit called for 29 data bytes
[22:50:44][VV][elero:151]: wait_rx
[22:50:44][VV][elero:177]: wait_tx
[22:50:44][VV][elero:191]: wait_tx_done
[22:50:44][V][elero:231]: Transmission successful
```

How the received message looks like (STOP):

```
[23:03:24][VV][elero:015]: loop says "received"
[23:03:24][D][elero:451]: rcv'd: len=29, cnt=114, typ=0x6a, typ2=0x10, hop=0a, syst=01, chl=17, src=0x82a501, bwd=0x82a501, fwd=0x82a501, #dst=01, dst=a46d20, rssi=-58.0, lqi=42, crc= 1, payload=[0x00 0x03 0x00 0x00 0x10 0x00 0x00 0x00 0x00 0x40]
```

Could this be the problem why sending the commands is not working?

andyboeh, 2024-10-27:
Please post the YAML configuration of this blind, it's hard to say without knowing what you configured.

sidious38, 2024-10-27:
Thank you for your fast response!

```
external_components:
  - source: github://andyboeh/esphome-elero

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO19

elero:
  cs_pin: GPIO5
  gdo0_pin: GPIO14

cover:
  - platform: elero
    blind_address: 0xa46d20
    channel: 17
    remote_address: 0x82a501
    name: Sommergarten
    open_duration: 25s
    close_duration: 22s
    poll_interval: 30min
    payload_1: 0x00
    payload_2: 0x03
    pck_inf1: 0x6a
    pck_inf2: 0x10
    hop: 0x0a
    ```

andyboeh, 2024-10-27:
Your configuration seems right to me. Regarding your questions: The log file contains the **decrypted** received commands, but the **encrypted** sent commands - that's why they differ so heavily. I agree that this is somewhat confusing.

In the Home Assistant forum, a user reported a similar problem with the same `payload_2` type (0x03) as yours (mine had all 0x04 and we have a mixture of RolTop, RolTop-J, RolMotion and Combio motors/receivers): https://community.home-assistant.io/t/elero-usb-transmitter-to-control-elero-and-weinor-products/52740/259. Maybe the protocol is slightly different on these blinds?

Do you know which motors you have exactly? And which remote control?

sidious38, 2024-10-31:
The remote is called BiEasy-5M, which should correspond to a VarioTel-2 remote (the contents of the manual are exactly the same). I don't know anything about the motor, but it seems like they just rebranded an Elero motor for Weinor.

I've read in reviews of the VarioTel-2 remote that some people use it to control Weinor blinds, so I think it must be possible to reuse your library. I will try to analyse the learning process as it might be possible to learn the virtual remote as a second remote for the blinds. At first glance, the learning process didn't seem too complex when I analysed the messages, but I will have to adapt the source code. Maybe I'll be successful.

Thank you for the effort you have already put into this program!

andyboeh, 2024-12-19:
In the meantime, I did some tests with a Schlotterer-branded RolTop, but it's still working for me.

Did you try to simplify your configuration and stay with the defaults? Did it make a difference?

sidious38, 2024-12-28:
As I'm on holiday at the moment, I'll have another look at the controller next week. I will report back then.

sidious38, 2025-01-01:
It turns out that the address used by the remote to send signals to the blind differs from the address used by the blind to report its state back. Out of curiosity, I tested using the source address of the blind (the one it uses to send its state back), and this approach worked perfectly. I still had to use other payload and pck_inf bytes than you did.

Again, thank you for publishing this program, it will help us to keep our "Sommergarten" cooler in summer because we can no longer forget to do it manually :-)

andyboeh, 2025-01-01:
Great that you got it working! Does that mean that your blind has two addresses, one that is used a destination for your original remote and one that is used as destination for this system?

dipsk2, 2025-01-02:
Thanks @andyboeh for this amazing solution. I hop on to the conversation because I guess I am facing the same issue. My blinds are from brand "Griesser" which seems to use yet another re-branded Elero system -- also its manual looks the same.

I also have issues sending the signal. Receiving is fine and configuration seems to be correct. 

Logs when reading from remote:
```
rcv'd: len=29, cnt=71, typ=0x6a, typ2=0x10, hop=00, syst=01, chl=17, src=0xba5200, bwd=0xba5200, fwd=0xba5200, #dst=01, dst=85662b, rssi=-56.0, lqi=127, crc= 1, payload=[0x00 0x01 0x00 0x00 0x10 0x00 0x00 0x00 0x00 0x40]
```

Yaml
```
spi:
  clk_pin: GPIO33
  mosi_pin: GPIO14
  miso_pin: GPIO39

elero:
  cs_pin: GPIO12
  gdo0_pin: GPIO32

cover:
  - platform: elero
    blind_address: 0x85662b
    channel: 17
    remote_address: 0xba5200
    name: Room2  
    open_duration: 52s
    close_duration: 52s
    supports_tilt: False
    payload_1: 0x00
    payload_2: 0x01
    pck_inf1: 0x6a
    pck_inf2: 0x10
    hop: 00
    command_check: 0x10
    command_stop: 0x10
    command_up: 0x21
    command_down: 0x41
```

Logs when sending:
```
[VV][elero:474]: send_command called
[V][elero:504]: send: len=29, cnt=02, typ=0x6a, typ2=0x10, hop=00, syst=01, chl=17, src=0xba5200, bwd=0xba5200, fwd=0xba5200, #dst=01, dst=0x85662b, payload=[0x00 0x01 0x83 0xab 0x4f 0x50 0x79 0xa7 0xd3 0xad]
[VV][elero:206]: transmit called for 29 data bytes
[VV][elero:151]: wait_rx
[VV][elero:177]: wait_tx
[VV][elero:191]: wait_tx_done
[V][elero:231]: Transmission successful
[VV][elero:015]: loop says "received"
[VV][elero:474]: send_command called
[V][elero:504]: send: len=29, cnt=02, typ=0x6a, typ2=0x10, hop=00, syst=01, chl=17, src=0xba5200, bwd=0xba5200, fwd=0xba5200, #dst=01, dst=0x85662b, payload=[0x00 0x01 0x83 0xab 0x4f 0x50 0x79 0xa7 0xd3 0xad]
[VV][elero:206]: transmit called for 29 data bytes
[VV][elero:151]: wait_rx
[VV][elero:177]: wait_tx
[E][elero:186]: Timed out waiting for TX: 0x0d
[VV][elero:060]: flush_and_rx
[VV][elero:164]: wait_idle
```
(Nothing happens with the blinds even when I see transmission successful message)

May be it is the same case as @sidious38 where its may be using a different signal back. @sidious38 if you managed to make it work, would it be possible to share your solution here please?

sidious38, 2025-01-03:
@andyboeh That's exactly the point that I don't really understand yet. For example, the sequence of a DOWN signal looks like this for me. The remote first sends a packet with len=32 and #dst=02, all other parameters are the same. The destnation address (a46d20) is the one I tried before and it doesn't work. The blinds then respond with a status message (0xca), but this has a completely different source address. I have actually always ignored this message before. Then the normal packet is sent with len=29.

With the log level *Verbose* I then noticed that the ESP was trying to set the status for the blind `0xf54c26`, see `Got state: 0x01 for blind 0xf54c26`. I then thought, well, maybe at least the status will be recognised if I operate the blind with the remote control and set the parameter `blind_address` to `0xf54c26`. It actually worked and because I had the view open in Home Assistant, I also happened to press the UP/DOWN buttons and it worked. To be honest, I wasn't expecting that.

```
[10:58:47][D][elero:451]: rcv'd: len=32, cnt=244, typ=0x6a, typ2=0x10, hop=00, syst=01, chl=17, src=0x82a501, bwd=0x82a501, fwd=0x82a501, #dst=02, dst=a46d20, rssi=-77.0, lqi=46, crc= 1, payload=[0x00 0x03 0x00 0x00 0x40 0x00 0x00 0x00 0x00 0x40]
[10:58:47][VV][elero:015]: loop says "received"
[10:58:47][D][elero:451]: rcv'd: len=29, cnt=134, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=244, src=0xf54c26, bwd=0xf54c26, fwd=0x82a501, #dst=01, dst=82a501, rssi=-89.0, lqi=48, crc= 1, payload=[0x01 0x78 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x10]
[10:58:47][V][elero.cover:115]: Got state: 0x01 for blind 0xf54c26
[10:58:49][VV][elero:015]: loop says "received"
[10:58:49][D][elero:451]: rcv'd: len=29, cnt=246, typ=0x6a, typ2=0x10, hop=0a, syst=01, chl=17, src=0x82a501, bwd=0x82a501, fwd=0x82a501, #dst=01, dst=a46d20, rssi=-78.0, lqi=46, crc= 1, payload=[0x00 0x03 0x00 0x00 0x40 0x00 0x00 0x00 0x00 0x40]
```

sidious38, 2025-01-03:
@dipsk2 This is my setup for the status message `0xca` from the blind.

```
[10:58:47][D][elero:451]: rcv'd: len=29, cnt=134, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=244, src=0xf54c26, bwd=0xf54c26, fwd=0x82a501, #dst=01, dst=82a501, rssi=-89.0, lqi=48, crc= 1, payload=[0x01 0x78 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x10]
```

```
cover:
  - platform: elero
    blind_address: 0xf54c26
    channel: 17
    remote_address: 0x82a501
    name: Sommergarten
    poll_interval: 5min
    supports_tilt: False
    payload_1: 0x00
    payload_2: 0x03
    pck_inf1: 0x6a
    pck_inf2: 0x10
    hop: 0x0a
    command_check: 0x00
    command_stop: 0x10
    command_up: 0x20
    command_down: 0x40
    command_tilt: 0x24
``

closed 2025-01-25 by sidious38