andyboeh/esphome-elero #12: Blinds stop moving after setting a % position

permissionBRICK, 2025-01-01:
Hi,

first of all, thanks so much for making this. I set it up with an ESP32 and the cc1101 868mhz board from amazon and its working great. One small Issue I noticed is that whenever I set the blinds not to Up or Down or Stop, but to a specific position (where it uses the seconds timer to stop it in time I assume) then that works, it does stop at the correct position. The Problem is after: If I then use the regular remote to control the blinds instead of home assistant, the blinds will start moving and instantly stop again twice, only on the third or fourth try they will go all the way up or down.
This behavior persists many minutes after having used the % feature, the only way to restore original behavior with the remote is to unplug and replug the esp32, then the blind works again as intended until I set another % value.

Sounds like whatever timer / thread you're using to stop it in time gets stuck looping for that window (the others work fine until I also set a % for them), even after having successfully stopped the blind? Any way to add a maximum time limit there?

andyboeh, 2025-01-01:
Please post your config and the log file from when you set the position until the blind finally moves as expected using your remote control.

permissionBRICK, 2025-01-02:
[log.txt](https://github.com/user-attachments/files/18292035
[cover-wohnung.yaml.txt](https://github.com/user-attachments/files/18292046/cover-wohnung.yaml.txt)
/log.txt)
Not much to see I guess since the log does not log the sent commands but only the received ones?

Observed behavior in detail:

Send Remote up or down comands - works as normal
send Home Assistant up down commands - works as normal 
(also updates the status most of the time)

But as soon as I send a set positon % command: IT runs to that position and stops, but if I then send an up or down command from the old remote, it will sometimes stop instantly after starting to move.

If I then send an up / down command from home assistant, or unplug and re/plug the ESP, the behavior with the old remote is back to normal.

permissionBRICK, 2025-01-02:
I looked through the code and think I found the issue. -> this seems to fix it for me:
[PR](https://github.com/andyboeh/esphome-elero/pull/13)

closed 2025-01-04 by permissionBRICK