andyboeh/esphome-elero #15: Blinds Movement Issue

breinsp, 2025-01-12:
Hi,

I'm experiencing an issue where my blinds don't consistently open or close fully after a single command from the interface. There's roughly a 1/3 chance they will work as expected. Most of the time, the motor gives a short jerk and stops immediately. To get them to move completely, I have to send the command multiple times.

Any insights into what could be causing this?

Thanks in advance for your help!

Here is my config
```
substitutions:
  name: esphome-web-2f0414
  friendly_name: Blinds

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  level: DEBUG

api:

ota:
  platform: esphome

improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

esp32_improv:
  authorizer: none

web_server:

external_components:
  - source: github://andyboeh/esphome-elero
    refresh: 0s

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO19

elero:
  cs_pin: GPIO5
  gdo0_pin: GPIO17
  freq0: 0x7a
  freq1: 0x71
  freq2: 0x21

cover:
  - platform: elero
    blind_address: 0x8f740f
    channel: 1
    remote_address: 0x86ab10
    name: Balcony
    payload_1: 0x00
    payload_2: 0x04
    pck_inf1: 0x6a
    pck_inf2: 0x00
    hop: 0x00
  - platform: elero
    blind_address: 0xf83a0f
    channel: 2
    remote_address: 0x86ab10
    name: Living Room
    payload_1: 0x00
    payload_2: 0x04
    pck_inf1: 0x6a
    pck_inf2: 0x00
    hop: 0x00
  - platform: elero
    blind_address: 0x1b730f
    channel: 3
    remote_address: 0x86ab10
    name: Bedroom
    payload_1: 0x00
    payload_2: 0x04
    pck_inf1: 0x6a
    pck_inf2: 0x00
    hop: 0x00
```

andyboeh, 2025-01-12:
Please provide the log file from both, a working and a failed attempt.

breinsp, 2025-01-12:
Blinds position at the middle, command "DOWN" followed by a short jerky motion but stopping immediately:

```
[21:44:52][D][cover:076]: 'Bedroom' - Setting
[21:44:52][D][cover:086]:   Command: CLOSE
[21:44:52][V][elero.cover:220]: Sending CLOSE command
[21:44:52][D][cover:170]: 'Bedroom' - Publishing:
[21:44:52][D][cover:178]:   State: CLOSED
[21:44:52][D][cover:186]:   Current Operation: CLOSING
[21:44:52][VV][api.service:086]: send_cover_state_response: CoverStateResponse {
  key: 3788924735
  legacy_state: LEGACY_COVER_STATE_CLOSED
  position: 0
  tilt: 0
  current_operation: COVER_OPERATION_IS_CLOSING
}
[21:44:52][W][component:237]: Component api took a long time for an operation (67 ms).
[21:44:52][W][component:238]: Components should block for at most 30 ms.
[21:44:52][VV][elero:474]: send_command called
[21:44:52][V][elero:504]: send: len=29, cnt=17, typ=0x6a, typ2=0x00, hop=00, syst=01, chl=03, src=0x86ab10, bwd=0x86ab10, fwd=0x86ab10, #dst=01, dst=0x1b730f, payload=[0x00 0x04 0x5f 0xe4 0x53 0x2c 0x68 0x4e 0xaf 0xe2]
[21:44:52][VV][elero:206]: transmit called for 29 data bytes
[21:44:52][VV][elero:151]: wait_rx
[21:44:52][VV][elero:177]: wait_tx
[21:44:52][VV][elero:191]: wait_tx_done
[21:44:52][V][elero:231]: Transmission successful
[21:44:52][VV][elero:015]: loop says "received"
[21:44:52][VV][elero:474]: send_command called
[21:44:52][V][elero:504]: send: len=29, cnt=17, typ=0x6a, typ2=0x00, hop=00, syst=01, chl=03, src=0x86ab10, bwd=0x86ab10, fwd=0x86ab10, #dst=01, dst=0x1b730f, payload=[0x00 0x04 0x5f 0xe4 0x53 0x2c 0x68 0x4e 0xaf 0xe2]
[21:44:52][VV][elero:206]: transmit called for 29 data bytes
[21:44:52][VV][elero:151]: wait_rx
[21:44:52][VV][elero:177]: wait_tx
[21:44:52][VV][elero:191]: wait_tx_done
[21:44:52][V][elero:231]: Transmission successful
[21:44:52][VV][elero:015]: loop says "received"
[21:44:52][VV][elero:474]: send_command called
[21:44:52][V][elero:504]: send: len=29, cnt=18, typ=0x6a, typ2=0x00, hop=00, syst=01, chl=03, src=0x86ab10, bwd=0x86ab10, fwd=0x86ab10, #dst=01, dst=0x1b730f, payload=[0x00 0x04 0x81 0x3b 0x06 0x70 0x71 0x07 0xd5 0xbd]
[21:44:52][VV][elero:206]: transmit called for 29 data bytes
[21:44:52][VV][elero:151]: wait_rx
[21:44:52][VV][elero:177]: wait_tx
[21:44:52][VV][elero:191]: wait_tx_done
[21:44:52][V][elero:231]: Transmission successful
[21:44:52][VV][elero:015]: loop says "received"
[21:44:52][VV][elero:474]: send_command called
[21:44:52][V][elero:504]: send: len=29, cnt=18, typ=0x6a, typ2=0x00, hop=00, syst=01, chl=03, src=0x86ab10, bwd=0x86ab10, fwd=0x86ab10, #dst=01, dst=0x1b730f, payload=[0x00 0x04 0x81 0x3b 0x06 0x70 0x71 0x07 0xd5 0xbd]
[21:44:52][VV][elero:206]: transmit called for 29 data bytes
[21:44:52][VV][elero:151]: wait_rx
[21:44:52][VV][elero:177]: wait_tx
[21:44:52][VV][elero:191]: wait_tx_done
[21:44:52][V][elero:231]: Transmission successful
[21:44:52][W][component:237]: Component elero.cover took a long time for an operation (51 ms).
[21:44:52][W][component:238]: Components should block for at most 30 ms.
[21:44:52][VV][elero:015]: loop says "received"
[21:44:52][VV][elero:015]: loop says "received"
[21:44:52][D][elero:451]: rcv'd: len=29, cnt=20, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=17, src=0x1b730f, bwd=0x1b730f, fwd=0x86ab10, #dst=01, dst=86ab10, rssi=-87.0, lqi=47, crc= 1, payload=[0x01 0x62 0x00 0x00 0x00 0x00 0x00 0x00 0x0b 0x10]
[21:44:52][V][elero.cover:116]: Got state: 0x0b for blind 0x1b730f
[21:44:53][VV][elero:015]: loop says "received"
[21:44:53][D][elero:451]: rcv'd: len=29, cnt=21, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=18, src=0x1b730f, bwd=0x1b730f, fwd=0x86ab10, #dst=01, dst=86ab10, rssi=-85.0, lqi=47, crc= 1, payload=[0x01 0x62 0x00 0x00 0x00 0x00 0x00 0x00 0x0d 0x90]
[21:44:53][V][elero.cover:116]: Got state: 0x0d for blind 0x1b730f
[21:44:53][D][cover:170]: 'Bedroom' - Publishing:
[21:44:53][D][cover:178]:   State: CLOSED
[21:44:53][D][cover:186]:   Current Operation: IDLE
[21:44:53][VV][api.service:086]: send_cover_state_response: CoverStateResponse {
  key: 3788924735
  legacy_state: LEGACY_COVER_STATE_CLOSED
  position: 0
  tilt: 0
  current_operation: COVER_OPERATION_IDLE
}
[21:44:53][W][component:237]: Component elero took a long time for an operation (65 ms).
[21:44:53][W][component:238]: Components should block for at most 30 ms.
[21:44:54][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=200642 (now=210643)
[21:45:01][VV][api.connection:133]: Sending keepalive PING...
[21:45:01][VV][api.service:037]: send_ping_request: PingRequest {}
[21:45:01][VV][api.service:706]: on_ping_response: PingResponse {}
[21:45:04][VV][elero:015]: loop says "received"
```

breinsp, 2025-01-12:
Blinds position at the middle, command "DOWN" spammed a few times until it starts moving to fully closed position:

```
[21:54:32][D][cover:170]: 'Bedroom' - Publishing:
[21:54:32][D][cover:178]:   State: CLOSED
[21:54:32][D][cover:186]:   Current Operation: IDLE
[21:54:32][VV][api.service:086]: send_cover_state_response: CoverStateResponse {
  key: 3788924735
  legacy_state: LEGACY_COVER_STATE_CLOSED
  position: 0
  tilt: 0
  current_operation: COVER_OPERATION_IDLE
}
[21:54:32][W][component:237]: Component elero took a long time for an operation (73 ms).
[21:54:32][W][component:238]: Components should block for at most 30 ms.
[21:54:32][VV][api.service:752]: on_cover_command_request: CoverCommandRequest {
  key: 3788924735
  has_legacy_command: NO
  legacy_command: LEGACY_COVER_COMMAND_OPEN
  has_position: YES
  position: 0
  has_tilt: NO
  tilt: 0
  stop: NO
}
[21:54:32][D][cover:076]: 'Bedroom' - Setting
[21:54:32][D][cover:086]:   Command: CLOSE
[21:54:32][V][elero.cover:220]: Sending CLOSE command
[21:54:32][D][cover:170]: 'Bedroom' - Publishing:
[21:54:32][D][cover:178]:   State: CLOSED
[21:54:32][D][cover:186]:   Current Operation: CLOSING
[21:54:32][VV][api.service:086]: send_cover_state_response: CoverStateResponse {
  key: 3788924735
  legacy_state: LEGACY_COVER_STATE_CLOSED
  position: 0
  tilt: 0
  current_operation: COVER_OPERATION_IS_CLOSING
}
[21:54:32][W][component:237]: Component api took a long time for an operation (65 ms).
[21:54:32][W][component:238]: Components should block for at most 30 ms.
[21:54:32][VV][elero:474]: send_command called
[21:54:32][V][elero:504]: send: len=29, cnt=131, typ=0x6a, typ2=0x00, hop=00, syst=01, chl=03, src=0x86ab10, bwd=0x86ab10, fwd=0x86ab10, #dst=01, dst=0x1b730f, payload=[0x00 0x04 0xef 0x06 0xc3 0xfa 0x28 0x95 0xef 0x81]
[21:54:33][VV][elero:206]: transmit called for 29 data bytes
[21:54:33][VV][elero:151]: wait_rx
[21:54:33][VV][elero:177]: wait_tx
[21:54:33][VV][elero:191]: wait_tx_done
[21:54:33][V][elero:231]: Transmission successful
[21:54:33][W][component:237]: Component elero.cover took a long time for an operation (54 ms).
[21:54:33][W][component:238]: Components should block for at most 30 ms.
[21:54:33][VV][elero:015]: loop says "received"
[21:54:33][VV][elero:474]: send_command called
[21:54:33][V][elero:504]: send: len=29, cnt=131, typ=0x6a, typ2=0x00, hop=00, syst=01, chl=03, src=0x86ab10, bwd=0x86ab10, fwd=0x86ab10, #dst=01, dst=0x1b730f, payload=[0x00 0x04 0xef 0x06 0xc3 0xfa 0x28 0x95 0xef 0x81]
[21:54:33][VV][elero:206]: transmit called for 29 data bytes
[21:54:33][VV][elero:151]: wait_rx
[21:54:33][VV][elero:177]: wait_tx
[21:54:33][E][elero:186]: Timed out waiting for TX: 0x0d
[21:54:33][VV][elero:060]: flush_and_rx
[21:54:33][VV][elero:164]: wait_idle
[21:54:33][D][elero.cover:087]: Retry #0 for blind 0x1b730f
[21:54:33][W][component:237]: Component elero.cover took a long time for an operation (94 ms).
[21:54:33][W][component:238]: Components should block for at most 30 ms.
[21:54:33][VV][elero:015]: loop says "received"
[21:54:33][D][elero:451]: rcv'd: len=29, cnt=242, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=131, src=0x1b730f, bwd=0x1b730f, fwd=0x86ab10, #dst=01, dst=86ab10, rssi=-90.0, lqi=52, crc= 1, payload=[0x01 0x62 0x00 0x00 0x00 0x00 0x00 0x00 0x0d 0x10]
[21:54:33][V][elero.cover:116]: Got state: 0x0d for blind 0x1b730f
[21:54:33][D][cover:170]: 'Bedroom' - Publishing:
[21:54:33][D][cover:178]:   State: CLOSED
[21:54:33][D][cover:186]:   Current Operation: IDLE
[21:54:33][VV][api.service:086]: send_cover_state_response: CoverStateResponse {
  key: 3788924735
  legacy_state: LEGACY_COVER_STATE_CLOSED
[... 21 lines cut ...]
[21:54:47][VV][elero:015]: loop says "received"
[21:54:47][D][elero:451]: rcv'd: len=27, cnt=154, typ=0x45, typ2=0x00, hop=aa, syst=01, chl=01, src=0x39fa03, bwd=0x34730f, fwd=0x39fa03, #dst=01, dst=000001, rssi=-89.0, lqi=58, crc= 1, payload=[0x00 0x43 0x00 0x00 0x00 0x06 0x10 0x00 0x01 0x30]
[21:54:47][VV][elero:015]: loop says "received"
[21:54:47][D][elero:451]: rcv'd: len=27, cnt=154, typ=0x45, typ2=0x00, hop=aa, syst=01, chl=01, src=0x39fa03, bwd=0x74740f, fwd=0x39fa03, #dst=01, dst=000001, rssi=-85.0, lqi=45, crc= 1, payload=[0x00 0x43 0x00 0x00 0x00 0x06 0x10 0x00 0x01 0x30]
[21:54:47][VV][elero:015]: loop says "received"
[21:54:47][D][elero:451]: rcv'd: len=27, cnt=154, typ=0x45, typ2=0x00, hop=aa, syst=01, chl=01, src=0x39fa03, bwd=0x75740f, fwd=0x39fa03, #dst=01, dst=000001, rssi=-87.0, lqi=50, crc= 1, payload=[0x00 0x43 0x00 0x00 0x00 0x06 0x10 0x00 0x01 0x30]
[21:54:54][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=800642 (now=810643)
[21:55:03][VV][elero:015]: loop says "received"
[21:55:03][D][elero:451]: rcv'd: len=27, cnt=250, typ=0x45, typ2=0x10, hop=05, syst=01, chl=01, src=0x73a40f, bwd=0x73a40f, fwd=0x73a40f, #dst=01, dst=000001, rssi=-87.0, lqi=47, crc= 1, payload=[0x00 0x47 0x00 0x00 0x00 0x00 0x37 0x00 0xff 0xa0]
[21:55:03][VV][elero:015]: loop says "received"
[21:55:03][D][elero:451]: rcv'd: len=27, cnt=250, typ=0x45, typ2=0x10, hop=15, syst=01, chl=01, src=0x73a40f, bwd=0x8f740f, fwd=0x73a40f, #dst=01, dst=000001, rssi=-80.0, lqi=48, crc= 1, payload=[0x00 0x47 0x00 0x00 0x00 0x00 0x37 0x00 0xff 0xa0]
[21:55:03][VV][elero:015]: loop says "received"
[21:55:04][VV][elero:015]: loop says "received"
[... cut ...]
[21:55:04][D][elero:451]: rcv'd: len=27, cnt=251, typ=0x45, typ2=0x10, hop=15, syst=01, chl=01, src=0x73a40f, bwd=0xbd720f, fwd=0x73a40f, #dst=01, dst=000001, rssi=-65.0, lqi=47, crc= 1, payload=[0x00 0x42 0x00 0x00 0x00 0x04 0x00 0x00 0x01 0x50]
[21:55:04][VV][elero:015]: loop says "received"
[21:55:04][D][elero:451]: rcv'd: len=27, cnt=251, typ=0x45, typ2=0x10, hop=25, syst=01, chl=01, src=0x73a40f, bwd=0x74740f, fwd=0x73a40f, #dst=01, dst=000001, rssi=-84.0, lqi=48, crc= 1, payload=[0x00 0x42 0x00 0x00 0x00 0x04 0x00 0x00 0x01 0x50]
[21:55:05][VV][elero:015]: loop says "received"
```

breinsp, 2025-01-12:
After further testing, it does not seem to be random - the system initially works as expected, but over time it enters a "corrupted state." In this state, sending a single command causes a brief attempt to move, but the motion stops almost immediately. To overcome this issue and get the system moving again, I have to repeatedly send the same command in quick succession. Waiting a few seconds between sending commands will always result in an immediate halt.

breinsp, 2025-01-12:
It almost feels like the blinds are telling the controller they are closed/open to the max even though they are not.

andyboeh, 2025-01-12:
Hm, I just decoded the two commands that are sent: 0x40 for down, 0x00 for check. Maybe your blind stops on check? These checks are not executed immediately after boot, so that would explain your problem. 
Unfortunately, I did not add a configuration option to disable polling. You could try setting `command_check` to something invalid (like `0x99` or so).

breinsp, 2025-01-13:
You are big 🧠 that did the trick - thanks a lot! Now I can control my neighbors blinds, awesome project!

closed 2025-01-13 by breinsp