andyboeh/esphome-elero #16: position issue in intermediate position

u20p17, 2025-01-12:
Hello!
i tried the latest version but it seems that there is a bug in the position controller: Iopened the blind completely and then send teh command to move to 50%. It started moving and at 50% the blind stopped moving, but in HomeAssistant (and also the esp log I could see that the percentage didn't stop and was going down until it was 0... 

```
[18:00:03][D][cover:076]: 'Schlafzimmer' - Setting
[18:00:03][D][cover:084]:   Position: 50%
[18:00:03][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:03][D][cover:173]:   Position: 100%
[18:00:03][D][cover:186]:   Current Operation: CLOSING
[18:00:03][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:03][D][cover:173]:   Position: 100%
[18:00:03][D][cover:186]:   Current Operation: CLOSING
[18:00:04][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:04][D][cover:173]:   Position: 96%
[18:00:04][D][cover:186]:   Current Operation: CLOSING
[18:00:05][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:05][D][cover:173]:   Position: 91%
[18:00:05][D][cover:186]:   Current Operation: CLOSING
[18:00:06][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:06][D][cover:173]:   Position: 87%
[18:00:06][D][cover:186]:   Current Operation: CLOSING
[18:00:07][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:07][D][cover:173]:   Position: 82%
[18:00:07][D][cover:186]:   Current Operation: CLOSING
[18:00:08][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:08][D][cover:173]:   Position: 78%
[18:00:08][D][cover:186]:   Current Operation: CLOSING
[18:00:09][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:09][D][cover:173]:   Position: 74%
[18:00:09][D][cover:186]:   Current Operation: CLOSING
[18:00:10][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:10][D][cover:173]:   Position: 69%
[18:00:10][D][cover:186]:   Current Operation: CLOSING
[18:00:11][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:11][D][cover:173]:   Position: 65%
[18:00:11][D][cover:186]:   Current Operation: CLOSING
[18:00:12][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:12][D][cover:173]:   Position: 61%
[18:00:12][D][cover:186]:   Current Operation: CLOSING
[18:00:13][D][elero:451]: rcv'd: len=29, cnt=07, typ=0xca, typ2=0x10, hop=1a, syst=01, chl=99, src=0xc7813a, bwd=0xeb813a, fwd=0x69aa24, #dst=01, dst=69aa24, rssi=-80.0, lqi=127, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x0b 0x10]
[18:00:13][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:13][D][cover:173]:   Position: 56%
[18:00:13][D][cover:186]:   Current Operation: CLOSING
[18:00:14][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:14][D][cover:173]:   Position: 52%
[18:00:14][D][cover:186]:   Current Operation: CLOSING
[18:00:15][D][elero:451]: rcv'd: len=29, cnt=08, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=02, src=0xc7813a, bwd=0xc7813a, fwd=0xb3fe00, #dst=01, dst=b3fe00, rssi=-68.0, lqi=127, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x0b 0x10]
[18:00:15][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:15][D][cover:173]:   Position: 50%
[18:00:15][D][cover:186]:   Current Operation: CLOSING
[18:00:15][D][sensor:094]: 'Wifi Signal': Sending state -60.00000 dBm with 0 decimals of accuracy
[18:00:15][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:15][D][cover:173]:   Position: 47%
[18:00:15][D][cover:186]:   Current Operation: CLOSING
[18:00:16][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:16][D][cover:173]:   Position: 43%
[18:00:16][D][cover:186]:   Current Operation: CLOSING
[18:00:18][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:18][D][cover:173]:   Position: 39%
[18:00:18][D][cover:186]:   Current Operation: CLOSING
[18:00:18][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:18][D][cover:173]:   Position: 34%
[18:00:18][D][cover:186]:   Current Operation: CLOSING
[18:00:19][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:19][D][cover:173]:   Position: 30%
[18:00:19][D][cover:186]:   Current Operation: CLOSING
[18:00:20][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:20][D][cover:173]:   Position: 25%
[18:00:20][D][cover:186]:   Current Operation: CLOSING
[18:00:21][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:21][D][cover:173]:   Position: 21%
[18:00:21][D][cover:186]:   Current Operation: CLOSING
[18:00:22][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:22][D][cover:173]:   Position: 17%
[18:00:22][D][cover:186]:   Current Operation: CLOSING
[18:00:23][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:23][D][cover:173]:   Position: 12%
[18:00:23][D][cover:186]:   Current Operation: CLOSING
[18:00:24][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:24][D][cover:173]:   Position: 8%
[18:00:24][D][cover:186]:   Current Operation: CLOSING
[18:00:25][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:25][D][cover:173]:   Position: 4%
[18:00:25][D][cover:186]:   Current Operation: CLOSING
[18:00:26][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:26][D][cover:173]:   Position: 0%
[18:00:27][D][cover:186]:   Current Operation: CLOSING
[18:00:27][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:27][D][cover:173]:   Position: 0%
[18:00:27][D][cover:186]:   Current Operation: CLOSING
[18:00:28][D][esp32.preferences:114]: Saving 1 preferences to flash...
[18:00:28][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[18:00:29][D][elero:451]: rcv'd: len=29, cnt=212, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=111, src=0x149c40, bwd=0x149c40, fwd=0x69aa24, #dst=01, dst=69aa24, rssi=-80.0, lqi=115, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x02 0x10]
[18:00:29][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:29][D][cover:173]:   Position: 0%
[18:00:29][D][cover:186]:   Current Operation: CLOSING
[18:00:29][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:29][D][cover:173]:   Position: 0%
[18:00:29][D][cover:186]:   Current Operation: CLOSING
[18:00:30][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:30][D][cover:173]:   Position: 0%
[18:00:30][D][cover:186]:   Current Operation: CLOSING
[18:00:31][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:31][D][cover:173]:   Position: 0%
[18:00:31][D][cover:186]:   Current Operation: CLOSING
[18:00:32][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:32][D][cover:173]:   Position: 0%
[18:00:32][D][cover:186]:   Current Operation: CLOSING
[18:00:33][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:33][D][cover:173]:   Position: 0%
[18:00:33][D][cover:186]:   Current Operation: CLOSING
[18:00:35][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:35][D][cover:173]:   Position: 0%
[18:00:35][D][cover:186]:   Current Operation: CLOSING
[18:00:35][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:37][D][cover:173]:   Position: 0%
[18:00:38][D][cover:186]:   Current Operation: CLOSING
[18:00:38][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:38][D][cover:173]:   Position: 0%
[18:00:38][D][cover:186]:   Current Operation: CLOSING
[18:00:38][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:38][D][cover:173]:   Position: 0%
[18:00:38][D][cover:186]:   Current Operation: CLOSING
[18:00:39][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:39][D][cover:173]:   Position: 0%
[18:00:39][D][cover:186]:   Current Operation: CLOSING
[18:00:40][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:40][D][cover:173]:   Position: 0%
[18:00:40][D][cover:186]:   Current Operation: CLOSING
[18:00:40][D][sensor:094]: 'Uptime Sensor': Sending state 55.77600 s with 0 decimals of accuracy
[18:00:41][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:41][D][cover:173]:   Position: 0%
[18:00:41][D][cover:186]:   Current Operation: CLOSING
[18:00:41][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:42][D][cover:173]:   Position: 0%
[18:00:42][D][cover:186]:   Current Operation: CLOSING
[18:00:43][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:43][D][cover:173]:   Position: 0%
[18:00:43][D][cover:186]:   Current Operation: CLOSING
[18:00:43][D][elero:451]: rcv'd: len=29, cnt=11, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=116, src=0xc7813a, bwd=0xc7813a, fwd=0xeb813a, #dst=01, dst=69aa24, rssi=-69.0, lqi=127, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x0d 0x10]
[18:00:43][D][cover:170]: 'Schlafzimmer' - Publishing:
[18:00:43][D][cover:173]:   Position: 0%
[18:00:43][D][cover:186]:   Current Operation: IDLE
```

Any idea what could be the problem?

another question: how does the esp know in the very beginning the position of the blinds? shouldn't it move e.g. completely up/down (and measure the needed time) and then drive back the same time to be on the same position again?

Thanks!

andyboeh, 2025-01-12:
This is because your blind reports that it is still moving down. Only the very last message at [18:00:43] indicates that the blind has stopped moving. What happens is the following:

You request 50%, the blind starts moving. Upon reaching the desired position, the stop command is sent. Unfortunately, the blind reports at about the same time (obviously, after the stop command was sent) that it is still moving. Hence the ESP thinks it's still moving and counts along until it receives the stop message. 

It seems that your ESP doesn't reach your blind directly, at least all answers received were forwarded by other blinds - and that's the reason why they arrive so late. Moving your ESP nearer to the blind might help. I have no good idea how to solve this in code.

To your other question: It doesn't. You have to move the blind manually fully up or down once so that the position is in sync. I had thought about doing a full open first, but that would drive me crazy if my blinds moved like this.

That said, the position is just a rough estimate anyway. Blinds that are far away sometimes take several seconds until they respond to a command, but it's not consistent. This happens with the original remote and the transmitter stick as well.

u20p17, 2025-01-12:
thank you for your fast answer :) the esp was in the same room as the blind - maybe i have to try different frequency values… but how can i figure it out and not only try and error ? ;) what rrsi values you have?

andyboeh, 2025-01-12:
Yeah, different frequencies might work. 
Idk, I replaced all wireless blinds by wired ones.

u20p17, 2025-01-13:
I tried today again moving the esp nearer to the blind (as a test) and it changed the behaviour a little bit:

Blind was fully open --> I let it move to 89%. The blind stopped as expected. After that I moved to 71% and the blind stopped again, but it was showing  'Current Operation: CLOSING'. Compared to yesterday the % value in HA also stopped at 71%... After that I moved to 50% and everything was exactly the same as the step before. Same when moving upwards --> blind stops at x% and HA also shows x%, but it still shows ' Current Operation: OPENING' even it has already stopped...

here the log

```
[11:16:25][D][cover:170]: 'Bürotür' - Publishing:
[11:16:25][D][cover:173]:   Position: 100%
[11:16:25][D][cover:186]:   Current Operation: IDLE
[11:16:27][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[11:16:27][D][esp32.preferences:114]: Saving 2 preferences to flash...
[11:16:27][D][esp32.preferences:143]: Saving 2 preferences to flash: 0 cached, 2 written, 0 failed
[11:16:31][D][sensor:094]: 'Wifi Signal': Sending state -64.00000 dBm with 0 decimals of accuracy
[11:16:37][D][elero:451]: rcv'd: len=29, cnt=20, typ=0xca, typ2=0x10, hop=1a, syst=01, chl=66, src=0xc7813a, bwd=0xeb813a, fwd=0xf5813a, #dst=01, dst=69aa24, rssi=-75.0, lqi=127, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x90]
[11:16:38][D][elero:451]: rcv'd: len=29, cnt=112, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=72, src=0x3b9c40, bwd=0x3b9c40, fwd=0x69aa24, #dst=01, dst=69aa24, rssi=-80.0, lqi=127, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x02 0x90]
[11:16:40][D][cover:076]: 'Bürotür' - Setting
[11:16:40][D][cover:084]:   Position: 89%
[11:16:40][D][cover:170]: 'Bürotür' - Publishing:
[11:16:40][D][cover:173]:   Position: 100%
[11:16:40][D][cover:186]:   Current Operation: CLOSING
[11:16:40][D][cover:170]: 'Bürotür' - Publishing:
[11:16:40][D][cover:173]:   Position: 100%
[11:16:40][D][cover:186]:   Current Operation: CLOSING
[11:16:40][D][elero:451]: rcv'd: len=29, cnt=228, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=03, src=0x14823a, bwd=0x14823a, fwd=0xb3fe00, #dst=01, dst=b3fe00, rssi=-65.0, lqi=127, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x0b 0x10]
[11:16:41][D][cover:170]: 'Bürotür' - Publishing:
[11:16:41][D][cover:173]:   Position: 97%
[11:16:41][D][cover:186]:   Current Operation: CLOSING
[11:16:42][D][cover:170]: 'Bürotür' - Publishing:
[11:16:42][D][cover:173]:   Position: 95%
[11:16:42][D][cover:186]:   Current Operation: CLOSING
[11:16:43][D][cover:170]: 'Bürotür' - Publishing:
[11:16:43][D][cover:173]:   Position: 93%
[11:16:43][D][cover:186]:   Current Operation: CLOSING
[11:16:44][D][cover:170]: 'Bürotür' - Publishing:
[11:16:44][D][cover:173]:   Position: 90%
[11:16:44][D][cover:186]:   Current Operation: CLOSING
[11:16:45][D][elero:451]: rcv'd: len=29, cnt=230, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=04, src=0x14823a, bwd=0x14823a, fwd=0xb3fe00, #dst=01, dst=b3fe00, rssi=-63.0, lqi=127, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x0b 0x10]
[11:16:45][D][cover:170]: 'Bürotür' - Publishing:
[11:16:45][D][cover:173]:   Position: 89%
[11:16:45][D][cover:186]:   Current Operation: CLOSING
[11:16:45][D][elero:451]: rcv'd: len=29, cnt=231, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=04, src=0x14823a, bwd=0x14823a, fwd=0xb3fe00, #dst=01, dst=b3fe00, rssi=-64.0, lqi=127, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x0d 0x90]
[11:16:45][D][cover:170]: 'Bürotür' - Publishing:
[11:16:45][D][cover:173]:   Position: 88%
[11:16:45][D][cover:186]:   Current Operation: IDLE
[11:16:52][D][elero:451]: rcv'd: len=29, cnt=21, typ=0xca, typ2=0x10, hop=2a, syst=01, chl=74, src=0xc7813a, bwd=0xf5813a, fwd=0x69aa24, #dst=01, dst=69aa24, rssi=-80.0, lqi=113, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x90]
[11:16:52][D][elero:451]: rcv'd: len=29, cnt=02, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=78, src=0xf5813a, bwd=0xf5813a, fwd=0x69aa24, #dst=01, dst=69aa24, rssi=-80.0, lqi=104, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x90]
[11:16:52][D][elero:451]: rcv'd: len=29, cnt=113, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=80, src=0x3b9c40, bwd=0x3b9c40, fwd=0x69aa24, #dst=01, dst=69aa24, rssi=-80.0, lqi=127, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x02 0x90]
[11:17:00][D][cover:076]: 'Bürotür' - Setting
[11:17:00][D][cover:084]:   Position: 71%
[11:17:00][D][cover:170]: 'Bürotür' - Publishing:
[11:17:00][D][cover:173]:   Position: 88%
[11:17:00][D][cover:186]:   Current Operation: CLOSING
[11:17:00][D][cover:170]: 'Bürotür' - Publishing:
[11:17:00][D][cover:173]:   Position: 88%
[11:17:00][D][cover:186]:   Current Operation: CLOSING
[11:17:00][D][elero:451]: rcv'd: len=29, cnt=233, typ=0xca, typ2=0x10, hop=0a, syst=01, chl=05, src=0x14823a, bwd=0x14823a, fwd=0xb3fe00, #dst=01, dst=b3fe00, rssi=-66.0, lqi=127, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x0b 0x90]
[11:17:01][D][sensor:094]: 'Wifi Signal': Sending state -64.00000 dBm with 0 decimals of accuracy
[11:17:01][D][cover:170]: 'Bürotür' - Publishing:
[11:17:01][D][cover:173]:   Position: 86%
[11:17:01][D][cover:186]:   Current Operation: CLOSING
[11:17:01][D][esp32.preferences:114]: Saving 1 preferences to flash...
[11:17:01][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[11:17:02][D][cover:170]: 'Bürotür' - Publishing:
[11:17:02][D][cover:173]:   Position: 83%
[... cut ...]
[11:18:56][D][cover:186]:   Current Operation: IDLE
[11:19:01][D][sensor:094]: 'Wifi Signal': Sending state -64.00000 dBm with 0 decimals of accuracy
[11:19:01][D][esp32.preferences:114]: Saving 1 preferences to flash...
[11:19:01][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed

```

I have to find out how I can measure the frequency of my remote control as cheap as possible and I am also thinking of changing the wire antenna to a proper antenna with an sma connector... can you give me some hints there?

andyboeh, 2025-01-13:
I used a simple wire antenna. For measuring the frequency, a cheap RTL-SDR comes in handy (at least you should be able to compare with the original remote).

u20p17, 2025-01-15:
@andyboeh : today i mounted a bigger antenne (for 868MHz) on my board and it improved the signal a bit (lower rssi)... but i still had the same issue... Then i saw the last post in another issue #11 ... [sidious38](https://github.com/sidious38) last comment suggested to use 0a as hop (to be honest i have no idea why...) and suddenly everything seems to work... 😁  do you have an idea why?

sidious38, 2025-01-15:
> EDIT: I just recognized that in my last logs the hop was always 0a... hmm... but I am very sure that in the beginning when I check all settings I got 00 as hop... nevertheless - seems to work now...

@u20p17  I also see the hop=00 messages, but those are the ones I ignore because they are len=32.

andyboeh, 2025-01-16:
I have a basic idea, yes: `hop = 0x00` means that the command is sent directly to the blind, without any other blind relaying it (I'm sure about this). Hence, this only works if you are near enough the blind. 
Any other value is relayed by other blinds until it reaches the target blind. In order to avoid endless loops, the hop value is modified by each blind until it reaches zero (I'm not sure about this part, but it sounds reasonable). Unfortunately, this also leads to long delays - up to several seconds until the blind starts moving. I've seen this delay with the Elero Transmitter Stick as well when I had it mounted in the basement, so that's not a problem with the implementation but "working as designed".

I also noticed that some original remotes send `0x00` first and `0x0a` a bit later. I did not observe this behavior consistently, so it might as well be that the remote has learned that the blind is farther away and has adjusted to using `0x0a` by default.

Regarding the frequency: Unfortunately, not all boards are manufactured equally and some of the cheap chinese boards have wrong capacitors soldered (this has been discussed at length in the Homematic community). I did find my values also mostly by trial and error. The ones configured by default are the values used by the original remote (I got them from a sniff of the i2c communication between the MCU in the remote and the CC1101 in the remote).