andyboeh/esphome-elero #17: timed out waiting for idle

mithri, 2025-02-18:
Hi,

I changed my CC1101 module from 433MHZ: https://www.amazon.de/Funkmodul-RUIZHI-%C3%9Cbertragungsmodul-Antennenanschluss-Kompatibel/dp/B0CPY5LWFY?th=1 to 868MHZ: https://www.amazon.de/dp/B0C7N678C1?ref=ppx_yo2ov_dt_b_fed_asin_title and now i got this error message:

[20:33:16][C][spi:064]: SPI bus:
[20:33:16][C][spi:065]:   CLK Pin: GPIO18
[20:33:16][C][spi:066]:   SDI Pin: GPIO19
[20:33:16][C][spi:067]:   SDO Pin: GPIO23
[20:33:16][C][spi:072]:   Using HW SPI: SPI
[20:33:16][E][elero:172]: Timed out waiting for Idle: 0xff
[20:33:16][W][component:237]: Component elero took a long time for an operation (54 ms).
[20:33:16][W][component:238]: Components should block for at most 30 ms.
[20:33:16][C][elero:046]: Elero Config: 
[20:33:16][C][elero.cover:012]: Elero Cover 'Hauswirtschaftsraum'
[20:33:16][C][elero.cover:012]:   Assumed State: YES
[20:33:16][C][elero.cover:012]: Elero Cover 'Büro1'
[20:33:16][C][elero.cover:012]:   Assumed State: YES
[20:33:17][E][elero:172]: Timed out waiting for Idle: 0xff
[20:33:17][W][component:237]: Component elero took a long time for an operation (53 ms).
[20:33:17][W][component:238]: Components should block for at most 30 ms.
[20:33:17][C][elero.cover:012]: Elero Cover 'Büro2'
[20:33:17][C][elero.cover:012]:   Assumed State: YES
[20:33:17][E][elero:172]: Timed out waiting for Idle: 0x00
[20:33:17][W][component:237]: Component elero took a long time for an operation (53 ms).
[20:33:17][W][component:238]: Components should block for at most 30 ms.
[20:33:17][C][elero.cover:012]: Elero Cover 'Wohnzimmer'
[20:33:17][C][elero.cover:012]:   Assumed State: YES
[20:33:17][E][elero:172]: Timed out waiting for Idle: 0x00
[20:33:17][W][component:237]: Component elero took a long time for an operation (53 ms).
[20:33:17][W][component:238]: Components should block for at most 30 ms.
[20:33:17][C][elero.cover:012]: Elero Cover 'Wohnzimmer2'
[20:33:17][C][elero.cover:012]:   Assumed State: YES
[20:33:17][E][elero:172]: Timed out waiting for Idle: 0xff
[20:33:17][W][component:237]: Component elero took a long time for an operation (52 ms).
[20:33:17][W][component:238]: Components should block for at most 30 ms.

With the first module everything is working fine, but the reception is bad. Can anyone help me? Or any idea for a different CC1101 module with better reception?

Thank you!

andyboeh, 2025-02-19:
Seems like the module does not respond at all. It's either a bad module or wrong cabling. The Pinout on the linked product page is from the top, not the bottom.

mithri, 2025-02-19:
Hi, thank you for the fast response. I already tried another module (mostly the same layout) from another manufactur. It has the same error logs. And I tried everycable combination there is. Nevertheless, thank you. I will try again. other things.