andyboeh/esphome-elero #19: [HELP] Compilation failing - Can't get it to work

Crafter-Y, 2025-05-13:
Hi there, 

idk what I am doing wrong, but I can't even get it to compile. I'm using esphome 2025.4.2 on HA OS on rapi5.

```yaml
external_components:
  - source: github://andyboeh/esphome-elero@main

esphome:
  name: b-rolladen-steuerung
  friendly_name: B Rolladen Steuerung

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "REDACTED"

ota:
  - platform: esphome
    password: "REDACTED"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "B-Rolladen-Steuerung"
    password: "REDACTED"

captive_portal:

spi:
  id: spi_bus
  clk_pin: GPIO4
  miso_pin: GPIO5
  mosi_pin: GPIO6

elero:
  cs_pin: GPIO7
  gdo0_pin: GPIO3
  freq0: 0x7a         # Frequenzregister 0 (Standard 0x7a für ~868,3 MHz)
  freq1: 0x71
  freq2: 0x21

```

```
INFO ESPHome 2025.4.2
INFO Reading configuration /config/esphome/bastian-rolladen-steuerung.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing b-rolladen-steuerung (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
|-- SPI @ 2.0.0
Compiling .pioenvs/b-rolladen-steuerung/src/esphome/components/elero/elero.cpp.o
Compiling .pioenvs/b-rolladen-steuerung/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/b-rolladen-steuerung/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/b-rolladen-steuerung/src/esphome/components/mdns/mdns_libretiny.cpp.o
src/esphome/components/elero/elero.cpp:4:10: fatal error: esphome/components/elero/cover/EleroCover.h: No such file or directory
 #include "esphome/components/elero/cover/EleroCover.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pioenvs/b-rolladen-steuerung/src/esphome/components/mdns/mdns_rp2040.cpp.o
*** [.pioenvs/b-rolladen-steuerung/src/esphome/components/elero/elero.cpp.o] Error 1
========================== [FAILED] Took 2.19 seconds ==========================
```

Cleaning build files does not change the error.

I think that it might have something to do with one of the 2025 esphome updates.

andyboeh, 2025-05-13:
It's compiling fine on my 2025.4.1 ESPHome version. However, the ESP32-C3 was never tested and I'm pretty sure that this is the problem. If I change my platform to ESP32-C3, it fails with the same error. 

As can be seen on the start page, I'm not maintaining this anymore. If you fix it yourself and create a PR, I'm happy to apply it!

closed 2025-05-13 by andyboeh

Crafter-Y, 2025-05-13:
Okay, then that is definitely the issue. I'm pretty new to the esphome thing and thought esp32 is one thing. 

Thanks for taking your time here. I'll see if I can get this thing to work for me.

Crafter-Y, 2025-05-13:
The reason it didn't compile was, that I didn't include a cover in my yaml file. Because of that, the EleroCover didn't get loaded and the compile failed.

To solve the issue, I had to include a dummy cover.

andyboeh, 2025-05-14:
Great you found it. Now that you mention it: I ran into the same issue some time ago, and since then the need for a fake cover is written in bold on the front page ...