andyboeh/esphome-elero #22: LILYGO® T-embed ESP32-S3 cc1101 - Firmware compatible?

manuschillerdev, 2025-06-22:
Hi,

I am looking for an off-the-shelf Device that is compatible with this firmware so that I have a case, the antenna and a cc1101 Module at 868mhz without soldering.

One thing I found was the LILYGO® T-embed ESP32-S3 cc1101.

I'd expect the Display and Buttons to stop working when I Flash it with this firmware, which would be ok for me. Should the rest of the functionality work, though?

andyboeh, 2025-06-22:
Hi,

this is not a flashable firmware but a module for ESPHome. You need to compile the firmware yourself. Hence, there is no reason why Display or Buttons wouldn't work if you configure them properly.

See https://esphome.io/components/display/st7789v.html and https://esphome.io/components/binary_sensor/gpio.html

While I can't see a reason why the CC1101 shouldn't work, only testing will show proof. As this project is practically unmaintained, I'm not going to invest in such a device.

manuschillerdev, 2025-07-12:
In case anybody is wondering - it works just fine. Range is very good when configured to 868mhz.
I did not bother configuring anything else like the display or buttons, yet.

For me as someone who does not like soldering it is a nice out-of-the box experience together with this component.

```yaml
esphome:
  name: "lilygo-t-embed"
  platformio_options:
    board_build.mcu: esp32s3
    board_build.name: "LilyGO T-Embed ESP32-S3"
    board_build.upload.flash_size: "16MB"
    board_build.upload.maximum_size: 16777216
    board_build.vendor: "LilyGO"
    board_build.f_flash: 80000000L
    board_build.partitions: default_16MB.csv
    board_build.arduino.memory_type: qio_opi
  on_boot:
    then:
      - switch.turn_on: power_on
      # SW1:1  SW0:0 --- 315MHz
      # SW1:0  SW0:1 --- 868/915MHz
      # SW1:1  SW0:1 --- 434MHz
      - switch.turn_off: rf_sw1
      - switch.turn_on: rf_sw0

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: arduino
  flash_size: 16MB

wifi:
  networks:
    - ssid: ***
      password: ***

logger:
  level: DEBUG

api:

ota:
  platform: esphome

substitutions:
  friendly_name: "LilyGO T-Embed CC1101"

external_components:
  - source: github://andyboeh/esphome-elero

switch:
  - platform: gpio
    pin:
      number: GPIO15
      mode:
        output: True
    name: "${friendly_name} Power On"
    id: power_on
    restore_mode: ALWAYS_ON
  # SW1:1  SW0:0 --- 315MHz
  # SW1:0  SW0:1 --- 868/915MHz
  # SW1:1  SW0:1 --- 434MHz
  - platform: gpio
    pin: GPIO47
    id: rf_sw1
    restore_mode: ALWAYS_OFF
  - platform: gpio
    pin: GPIO48
    id: rf_sw0
    restore_mode: ALWAYS_ON

spi:
  clk_pin: GPIO11     # BOARD_SPI_SCK
  mosi_pin: GPIO9     # BOARD_SPI_MOSI
  miso_pin: GPIO10    # BOARD_SPI_MISO

elero:
  cs_pin: GPIO12      # BOARD_LORA_CS
  gdo0_pin: GPIO3     # BOARD_LORA_IO0
```

closed 2025-07-12 by manuschillerdev