andyboeh/esphome-elero #24: Problem with sending message

djipaul, 2025-08-14:
Hi @andyboeh,

First of all, thank you for creating this software! I'm currently in the process of setting it up for myself, but I'm unable to send any commands to the blind. Receiving works without any issues. I also checked the hardware and cabling, which all seem to be fine. Any help would be highly appreciated.

UP
`[17:14:39][D][elero:451]: rcv'd: len=29, cnt=42, typ=0x6a, typ2=0x10, hop=0x00, syst=0x01, chl=17, src=0xbb1b54, bwd=0xbb1b54, fwd=0xbb1b54, #dst=01, dst=0xaf3d42, rssi=-80.0, lqi=44, crc= 1, payload=[0x00 0x01 0x00 0x00 0x20 0x00 0x00 0x00 0x00 0xc0]
[17:14:39][VV][elero:015]: loop says "received`

DOWN
`[17:15:27][D][elero:451]: rcv'd: len=29, cnt=44, typ=0x6a, typ2=0x10, hop=0x00, syst=0x01, chl=17, src=0xbb1b54, bwd=0xbb1b54, fwd=0xbb1b54, #dst=01, dst=0xaf3d42, rssi=-79.0, lqi=45, crc= 1, payload=[0x00 0x01 0x00 0x00 0x40 0x00 0x00 0x00 0x00 0xc0]
[17:15:27][VV][elero:015]: loop says "received"`

STOP
`[17:16:17][D][elero:451]: rcv'd: len=29, cnt=49, typ=0x6a, typ2=0x10, hop=0x00, syst=0x01, chl=17, src=0xbb1b54, bwd=0xbb1b54, fwd=0xbb1b54, #dst=01, dst=0xaf3d42, rssi=-81.0, lqi=47, crc= 1, payload=[0x00 0x01 0x00 0x00 0x10 0x00 0x00 0x00 0x00 0xc0]
[17:16:17][VV][elero:015]: loop says "received"`

YAML
`external_components:
  - source: github://andyboeh/esphome-elero

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
  miso_pin: GPIO19

elero:
  cs_pin: GPIO5
  gdo0_pin: GPIO26

cover:
  - platform: elero
    blind_address: 0xaf3d42
    channel: 17
    remote_address: 0xbb1b54
    name: Arbeitszimmer
    payload_1: 0x00
    payload_2: 0x01
    pck_inf1: 0x6a
    pck_inf2: 0x10
    hop: 0x00
    command_stop: 0x10
    command_up: 0x20
    command_down: 0x40`

With this code, the blinds unfortunately don’t respond. After checking the logs, I noticed that sometimes the ESP receives some odd messages:

`[17:24:02][VV][elero:015]: loop says "received"
[17:24:02][D][elero:451]: rcv'd: len=29, cnt=113, typ=0xca, typ2=0x10, hop=0x0a, syst=0x01, chl=54, src=0xaf3d42, bwd=0xaf3d42, fwd=0xbb1b54, #dst=01, dst=0xbb1b54, rssi=-93.0, lqi=48, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x0d 0x10]`

`[17:24:42][VV][elero:015]: loop says "received"
[17:24:42][D][elero:451]: rcv'd: len=29, cnt=117, typ=0xca, typ2=0x10, hop=0x0a, syst=0x01, chl=58, src=0xaf3d42, bwd=0xaf3d42, fwd=0xbb1b54, #dst=01, dst=0xbb1b54, rssi=-95.0, lqi=50, crc= 1, payload=[0x01 0x60 0x00 0x00 0x00 0x00 0x00 0x00 0x0a 0x00]`

They all have 0xca as pck_inf1. I’m not sure why or when this happens, but I can reproduce these messages. I also tried changing the YAML to match these new messages, but unfortunately that didn’t work either.

Do you have any idea what could be wrong?

andyboeh, 2025-08-14:
Try to remove the "hop" parameter, maybe your signal needs to be relayed (0x00 means no hop, so you have to be very near the blind). It's possible that your transmitter is just not strong enough, but other blinds are going to relay the signal (this is what you see in your last three message, hop is 0x0a because the message is relayed; that's why src, bwd and fwd address are not equal).

Nobody except Elero knows exactly how the protocol works, so there is a lot of guesswork involved.

djipaul, 2025-08-17:
"Thank you for your answer! — I can confirm that this resolved the issue, and all the blinds are now working perfectly. Once again, a huge thank you for creating this software!"